SpenceKonde/ATTinyCore #500: Documentation: 404 in cross-reference to baud rates from the main documentation page.

prandeamus, 2021-01-27:
On the main page the paragraph starting 

"On the following chips, no hardware serial is available" ends with a link "See also the discussion of baud rates." This has URL https://github.com/SpenceKonde/ATTinyCore/blob/master/TinySoftSerialBaud.md which does not exist, leading to 404 error.

(Obviously not a blocker, but thought you would want to know)

SpenceKonde, 2021-01-27:
Fixed, thx - I think I may have once had extra docs in the root, but then moved to extras to tidy up the root of the core. I need a damned proofreader. Very good at writing shit down, very bad at not making a million topic run rambling sentence every so often, or sentences where wording midway through, Sometimes my sentences will just

closed 2021-01-27 by SpenceKonde

per1234, 2021-01-29:
@SpenceKonde would you like me to submit a PR to add a check for broken links in the Markdown files of the repository. https://github.com/tcort/markdown-link-check has been used for this purpose in the Arduino CLI repository for the last 8 months with good results. It's likely only a matter of adding a single .yml file to the repository. In the event there are links that are intentionally bad, a configuration file can be added to cause them to be ignored, but that's probably not needed here.

The only problem we've had is that sometimes you will get CI failures caused by temporary outages of the external websites, but that's not very frequent and I still think the benefits outweight the occasional minor inconveniences.

SpenceKonde, 2021-01-29:
YES Please :-)

I'm pretty sure I don't have any links that I want to go to error pages...

Really, I daresay I'd like to know if I'm linking to something that's down all the time so I can stop pointing my users there!