SpenceKonde/ATTinyCore #501: Question: TX only Serial where no HW Serial exists

prandeamus, 2021-01-27:
On parts that have no hardware serial (I am thinking here of Tiny88 but as you note there are others), the Serial object provides a version of SoftSerial that is tied to the AIN0/AIN1 gpios for TX and RX.

If my application only wishes to transmit and not receive serial data, am I free to use the TX pin and send data while configuring RX as an independent digital input i.e. on tiny88 would it be possible to transmit using PD6 and have a switch wired to PD7? There's no point in doing this if every time there's an input change on PD7(RX) that triggers an meaningless interrupt to the serial library.

In other words, if I use Serial and commit to using TX for output, are there any constraints on what inputs can attach to the unused RX pin?

pcfreak1201, 2021-01-27:
May be the same problem as in #443

SpenceKonde, 2021-01-27:
You may use the RX pin freely after following the procedure described on the part-specific doc for all parts that have the soft-serial-Serial thing: https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x8.md#uart-serial-support

SpenceKonde, 2021-01-27:
Seriously, people really gotta pay more attention to the part specific doc pages. Maybe not so much on the other two cores, but on ATTinyCore, everything is way different from everything else... Pre2016 AVRs are all kind of oddballs.... and the classic tinyAVRs are odder and ballier than their contempararies.

closed 2021-01-27 by SpenceKonde

prandeamus, 2021-01-27:
For avoidance of doubt, I'm just asking what is possible and to ensure that any limitations are understood. If code changes are needed to make a transmit-only Serial, I'm not going to throw my toys out of the pram or anything. I'll just look elsewhere (my current project is close to the limit of ATTiny88 I/O and I'm trying to determine if I can implement everything I want without switching to a more capable device or adding I/O extenders)

And as you say, these are oddball chips by modern standards. You're doing a great job of abstracting differences for devs, but I can imagine you're reaching the point of diminishing returns.