SpenceKonde/ATTinyCore #503: Attiny 1616 trouble in paradise

saysphilippe, 2021-02-01:
Hey!
We are trying to program the 1616 but it's just saying "uploading..." until it times out or I disconnect the USB to the programmer(Nano). I have a breadboard with both the 3216 and 1616 and the 3216 works perfectly fine. What can be wrong. The pinout are identical as well so I'm scratching my head here.

SpenceKonde, 2021-02-01:
Wrong repo, you want https://github.com/SpenceKonde/megaTinyCore which is what supports this. If it's jtag2updi, does pressing reset on the nano running jtag2updi during a hung upload cause it to work? I have had problems with older versions of jtag2updi - the latest one from my repo seems to work pretty well now.

saysphilippe, 2021-02-01:
Ah, sorry. Posted it under the wrong tree. I'm running the megaTinyCore of course. Let' me check this tomorrow. Thanks!

saysphilippe, 2021-02-02:
Then I downloaded the newest version of jtag2updi and have also tried to push the reset button on the Nano, but nothing works with the 1616. I tested also against the 3216 to check if it worked as it should and it does. Do you want me to  close this thread and open a new one under the right repo?

SpenceKonde, 2021-02-02:
Yeah, I think that would be the way to go

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

saysphilippe, 2021-02-02:
Closing and recreating under the right repo.

closed 2021-02-02 by saysphilippe