SpenceKonde/ATTinyCore #504: Another silicon bug on the ATtiny1634

felias-fogg, 2021-02-04:
I just noted another nasty feature of the PB2 bug on the ATtiny1634 that is not documented by Atmel. If you use PB2 as an output, set the output to high, have the watch dog disabled and put the CPU to sleep, then the MCU consumes 0.5 mA instead of 100 nA! So be aware of this "feature"! The following sketch demonstrates the behavior:

#include <avr/sleep.h>
void setup()
{
    pinMode(14, OUTPUT);
    digitalWrite(14, HIGH);
    ADCSRA = 0;
    set_sleep_mode(SLEEP_MODE_PWR_DOWN);
    sleep_mode();
 }

void loop()
{
}

SpenceKonde, 2021-02-06:
It's documented, kinda...They do say that the pin will be internally pulled down. And if a pin is being pulled down, while you drive it high, you'd have to expect that currrent would be flowing... but  yeah... it's a shame that this chip had to be saddled with those odd erratas. Those last 3 tinies seemed haphazard...  On the 841/441, the pullup enable isn't in the low I/O space, so you can't CBI/SBI... here it is, but there are several erratas that get in the way (classic avrs almost never had errata by the end, man - until these last tinies), Beyond that some things about them are just wacky. Why did the 841 get the epic differential ADC, which no arduino folk use, when the 828 has all the ADC channels?  Why make the awesome 841 oscillator, but then gimp it for the other two?

felias-fogg, 2021-02-07:
Yup, I noticed this later on. But since the errata was only warning against using the pin as an input, I was unsuspecting ... and then needed a few hours to find the problem. Well, I tracked it down to PB3 because of the errata, but was still puzzled  that you do something stupid like that and then never remove it in the following chip revisions. Once you know about, it is, of course, not a problem to program around it.

I did not try any of the newer Tinies since the 1634 is really quite what I need: a lot of pins for I/O (for building electronic geocaches that fit into a preform tube where there MCU has a lot of memory, can drive a dot matrix display and a few peripheral devices ;-)).

SpenceKonde, 2021-02-07:
Totally know a guy who had thousands of assembled devices by the time it
started to really bite. That's what happens when your PCB design guy
doesn't know the product line, designed the board and then left the scene
before you hired someone completely different to try to make the firmware
work.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

SpenceKonde, 2021-02-08:
The modern AVR parts are amazing. It feels weird to come back here into classic AVR land to work on ATTinyCore, tbh...Though the modern peripherals have crazy amount of errata :-P

closed 2021-02-08 by SpenceKonde