SpenceKonde/ATTinyCore #505: Micronucleus link broken in http://drazzy.com/package_drazzy.com_index.json

thomasonw, 2021-02-05:
The referenced file:  "https://github.com/ArminJo/DigistumpArduino/raw/master/tools/micronucleus-2.0a4-win.zip"
no longer exits.  It has been updated:  https://github.com/ArminJo/DigistumpArduino/tree/master/tools

SpenceKonde, 2021-02-05:
Sigh. See #465. When I have a chance to come back and make an interim fix (after DxCore 1.3.1 release), I will not make the same mistake again - the Micronucleus client binaries on which ATTinyCore depends will be hosted in a location which I control.

SpenceKonde, 2021-02-05:
I was able to locate a copy of the deleted windows package (note that there are two **DIFFERENT** windows binaries distributed online, _with the same filename_, and s_ame file size, down to the byte_. But the text of the .inf file is the same. Luckily I had a copy of the old one sitting in the staging subdirectory in one of my portable installations so I could re-release it with confidence that what I was releasing is the same

closed 2021-02-05 by SpenceKonde