SpenceKonde/ATTinyCore #507: Change in behaviour of USBtinyISP Programmer options

technoblogy, 2021-02-09:
I'm comparing the behaviour of ATTinyCore 1.4.0 and 1.4.1 on programming an ATtiny85 using USBTinyISP.
Settiings are ATtiny25/45/85 (No bootloader), ATtiny85, 1 MHz (internal):

#### 1.4.0

**USBTinyISP (ATTinyCore)** - works fine
**USBTinyISP SLOW (ATTinyCore)** - works fine

#### 1.4.1

The options are named differently:

**USBTinyISP FAST (ATTinyCore)** - fails with error:

````
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

An error occurred while uploading the sketch
````
**USBTinyISP SLOW (ATTinyCore)** - works fine

Is **USBTinyISP FAST (ATTinyCore)** what used to be called **USBTinyISP (ATTinyCore)**? If so, perhaps something has changed that's causing a problem.

SpenceKonde, 2021-02-09:
Okay I will try to document this; I remember typing about this, but forget where.

In 1.4.0 the two options might as well be named
USBtinyISP Almost as SLOW(ATTinyCore)
USBtinyISP SLOW (ATTinyCore) 

That is, there was virtually no difference between them, and both of them were dreadfully slow. I discovered this when , after rearranging the piles of crap on my desk, I found it awkward to connect the usbasps with the full size connector, and did some programming with the USBTinyISPs, and discovered that they took forever and a day to write. And that both of them were in practice slow enough to program at 1 MHz too. So I renamed the slightly-less-slow one to USBTinyISP FAST, and cranked up the speed to something that would be guaranteed to work at 8 MHz but not below that, and which wasn't so miserably slow to program.

SpenceKonde, 2021-02-09:
I don't even know where in the docs I would put that though, I don't have a section where I describe the programmers or anything

technoblogy, 2021-02-09:
Good - that makes sense. Perhaps some information about it here would be useful:

https://github.com/SpenceKonde/ATTinyCore/blob/master/Programming.md

I don't want to increase the size of the documentation unnecessarily, but it was a bit of a shock to me that the USBtinyISP option that always worked for me appeared to have suddenly stopped working, so I think it needs explaining. Thanks!

SpenceKonde, 2021-02-09:
Oh - huh - I actually did this at some point a couple months back... 

I really don't quite know why I didn't push out a release sooner... 
```
usbtinyisp.name=USBtinyISP FAST, for parts running >=2 MHz
usbtinyisp.protocol=usbtiny
usbtinyisp.program.tool=avrdude
usbtinyisp.program.extra_params=
usbtinyisp.program.speed=0.3
usbtinyisp.program.extra_params=-B{program.speed}

usbtinyisp2.name=USBtinyISP SLOW for <2 MHz incl new (1 MHz)
usbtinyisp2.protocol=usbtiny
usbtinyisp2.program.tool=avrdude
usbtinyisp2.program.speed=8
usbtinyisp2.program.extra_params=-B{program.speed}
```

closed 2021-02-09 by SpenceKonde

technoblogy, 2021-02-10:
OK, that's definitely better - thanks!

After thinking about it over night I was going to suggest that, because users assume that the first option on the list is the 'normal' one, the best solution is:

* Make USBtinyISP SLOW the first option and rename it USBtinyISP (as previously).

* Make USBtinyISP FAST the second option, implying that it's for special cases.

* Explain the difference in the documentation, but now it should still work for people who don't read the documentation.

By the way, in your new options what does "for <2 MHz incl new (1 MHz)" mean? It's not clear to me.

technoblogy, 2021-03-16:
Sorry, but this is still bugging me.

Because 1 MHz is the default on a new ATtiny, and you used to have to do a Burn Bootloader to change it to 8 MHz, most of my blog projects (and many other published projects) specified 1 MHz. People have got to know to choose the second menu option for these, or they get an unhelpful error. Why not make the dividing line 1 MHz instead of 2 MHz?

Also, in your proposed menu options I don't understand what **for <2 MHz incl new (1 MHz)** means.

So my suggestion is:

* Make **USBtinyISP FAST** the first option, make it work down to 1 MHz, and rename it **USBtinyISP** (as previously).
* Make **USBtinyISP SLOW** the second option and rename it **USBtinyISP SLOW for <1 MHz**.