SpenceKonde/ATTinyCore #509: When I use i2c communication, the servo starts bouncing.

bohumir-brocko, 2021-02-11:
I use Attiny481 and everything goes perfectly. Until I combined the two examples into one. I use the WireS and Servo library. I don't know why the pulse length for the servo starts to stretch. When I disconnect the communication, the servo runs nicely. Don't know where I'm making a mistake?

bohumir-brocko, 2021-02-11:
It will probably be a bigger problem, because I2c communication also disrupts delayMicroseconds. It just starts bouncing during i2c communication. I am not able to solve this on my level.

SpenceKonde, 2021-02-11:
Why the heck are we using WireS? That's the hardware I2C slave implementation? 

Does the included Wire library's slave implementation, with Tools -> Wire Mode set to Slave not work? 

I do not know what "starts bouncing" means in this context. Please explain with numbers, measurements, scope traces - something objective. 

Please see if it reproduces with the builtin Wire library in slave mode, and if it does, can you give me a minimal example that reproduces the problem? A test where the servo is set to one position and never changed, but which goes to hell the moment I2C communication is initiated would be a good test - easy to see on a scope without having to dig out a servo.

SpenceKonde, 2021-02-12:
(the included Wire library *has* an implementation of wire slave that uses the hardware TWI slave peripheral)

bohumir-brocko, 2021-02-12:
Yes you are right. I found some examples with WireS and it wasn't good. So I switched to Wire and everything is working. My question was probably confusing. I will continue to work on it. But it was a mystery. The error always occurred during ISR processing. When I entered the average value of 90 ° into the servo, the signal was correctly 1500us. During the i2c response, signals 10 - 20% longer suddenly appeared for a very short time. Sometimes the disorders lasted only a few us. They have always been in tactile i2c processing. I haven't seen that yet, so I panicked a bit. Maybe it won't show up again. Specifically, I combined the example of a virtual eeprom and a servo. I also added some other commands to the ISR processing and I probably shouldn't have done that. It was interesting that when I threw out the servo library and put only a pinwrite on 1500us, these signals also went bad.

SpenceKonde, 2021-02-12:
Glad to hear it is working! 

I think I understand what's going on here though... You've got stuff going on within an ISR - but you also need to keep outputting a stable servo signal. During an ISR, other interrupts are disabled (that's how classic AVRs work - you enter the ISR, and the I bit of SREG (global interrupt enable) is cleared by hardware, and set again when it hits the `reti` to exit the interrupt.) But how does servo do it's thing? By setting an interrupt on one of the timers (On classic AVR, I imagine this would be one of the 16-bit timers, also known as "the good timers" - of which the '841 is blessed with two of) - it's been ages since I've looked at the classic AVR servo library).... so if the wire interupt is happening when you need the falling edge of the servo pulse, and the duration of that delay is long enough that the servo can tell the difference, it's gonna mess up the servo behavior. 

Efficient ISRs will help greatly with this (otherwise you'd see it  from the millis interrupt - but you don't, because it's fast enough that the servo doesn't notice the difference in the pulse lengths); it sounds like that WireS library was particularly bad - but you can always get into dangerous territory when you're relying on something like Servo while running other potentially slow ISRs (one of the main reasons we don't like slow ISRs)... 

In this particular case, if circumstances get desperate... I think one could actually make the timer drive the servo output directly instead of using servo. You only get 2 servos, but if that's all you need, you're fine... you'd set CSn[2:0] = 0b010 for /8 prescale (either 1 or 2 counts per microsecond  depending on system clock), WGMn[3:0] 0b1110 (WGM 14, fast PWM, TOP = ICRn), ICRn = 19999 or 39999 (at 8 or 16 MHz respectively) Thus, it would be taking 20 ms per cycle (0-indexed, that is, it spends 1 tick at 0 and 1 tick at TOP). Then you could set OCRnA = (number of microseconds -1) (or ((2*number of us)-1) for 16 MHz) and poke the appropriate COMnA/B[1:0] bits and make sure PWM output was enabled (by setting the appropriate bit in TOCPMCOE (the pin mux output enable); 

I'd be inclined to find a pin already set up on the timer I wanted to use, set the CS of timer to 0 (stop the timer) analogWrite() the pin (setting the output mux and COM bits - but no pwm cause timer is stopped, and then reconfigure the pin, setting the CS bits last. 
Once that was done, you could change the length of the pulse just by writing output compare register, and it would not depend on interrupts!

SpenceKonde, 2021-02-12:
By the way, internally the hardware slave wire *is* based on a version of WireS. I remember universal wire involving a lot of library modification though...

SpenceKonde, 2021-02-12:
Note - I will still see and respond to messaes here - just clearing out tasks that don't need to be fixed in the core.

closed 2021-02-12 by SpenceKonde

bohumir-brocko, 2021-02-13:
Thank you for clarifying the timers. I've been playing with attins for only two months, so I haven't gone far enough to understand the entire documentation for this chip. Timers and their dependencies are still out of my focus. I just tried some examples and tried to do some experiments. I tried 85 and 84, but the sw implementation of the i2c slave turned out to be unusable. On recommendation, I tried 841 and it is excellent for an i2c slave. Thank you for taking the time to address this issue. I will try not to disturb the timing with my functions.