SpenceKonde/ATTinyCore #511: Unable to find pin mappings

mmiller7, 2021-02-14:
I'm unable to find the pin mappings for the ATTINY85, I know in previous versions of libraries there were multiple pinouts for the numbering.  My 1.8.13 IDE on 64-bit Linux does not appear to have a "Tools > Pin Mappings" menu to check as your documentation suggests.

Is there a default or manual way to select pin mappings that can be added to documentation?

pcfreak1201, 2021-02-14:
I'm not sure, if I ever saw different mappings at the t85, because there is no "official" Arduino t85, as I know.
On my Linux the mapping is at ~/arduino15/packages/ATTinyCore/hardware/avr/1.4.1/extras/Pinout_x5.jpg (as in the repository)

srw777, 2021-02-14:
Spence would have the best answer, but I seem to recall rumblings here
about removing the images from the installation package because they took
up so much space.  As long as you have internet, you can access the
documentation on the github site. Specifically, the T85 can be found here:
https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x5.md

ttyl
srw

SpenceKonde, 2021-02-14:
All parts with multiple pin mappings have a tools -> pin mapping submen. Parts which do not have multiple pin mappings do not have such an option. All of the pin mappings (default and alternate, where one exists) can be found in the part specific documentation pages in the repo. Ex: https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x5.md for the tiny85. 

As you can see, the tiny85 only has one pin mapping. 
All the pins are on the same port, and it is just so obviously favorable to number the pins in the same order that the manufacturer put them on the underlying hardware, that not even the living brain donors who came up with pin mappings for third partry hardware packages before I showed up (I claim authority to trash-talk them because I'm the one maintaining the messes they left behind, after theyported the core ; if you think I'm exaggerating how bad it is, look at the tiny167). The worst part about pinmappings is when I was looking at a few of the parts I support, and realizing that the pin mapping is irredemably bad, and I'd been keeping the core on life support for years without doing anything about the awful decisions that were made....

@srw777 - the thing about removing things from the packages only applies to the board manager versions (the past few versions when I was talking about it, who the hell knows! But I've now got a process, and the board manager package is made by a little shell script (even the json file entry!) so the deal with this is now very consistent: Everything will always be present in the manual downloads from 1.5.0 onwards (and 2.2.0 of megaTinyCore and 1.3.0 of DxCore). Board manager packages (same version considerations apply) do not have the extras folder present at all, on the reasoning that nobody is is going to install via board manager, then search through the C:/user/local/arduino15 subdirectories looking for the documentation, and probably the only person viewing them outside of a web browser is me when I write then in sublime text.. Though I probably should  also have it regex-find-replace  links from the README.md with links to the page on github instead of the

SpenceKonde, 2021-02-14:
Closing because this does not represent a defect in the core.

closed 2021-02-14 by SpenceKonde