SpenceKonde/ATTinyCore #516: DigiKeyboard doesn't work

philibertc, 2021-02-18:
Hello, I've been using an mh-tiny88 board for quite some time, it's a perfect board for classic projects, without serial port. I saw that it was possible to use it as a HID device, I created this code with the Digispark Keyboard library (or DigiKeyboard). Here is the code :
```c++
#include "DigiKeyboard.h"

void setup() {
  // don't need to set anything up to use DigiKeyboard
}

void loop() {
  // this is generally not necessary but with some older systems it seems to
  // prevent missing the first character after a delay:
  DigiKeyboard.sendKeyStroke(0);
 
  // Type out this string letter by letter on the computer (assumes US-style
  // keyboard)
  DigiKeyboard.println("Hello World!");
 
  // It's better to use DigiKeyboard.delay() over the regular Arduino delay()
  // if doing keyboard stuff because it keeps talking to the computer to make
  // sure the computer knows the keyboard is alive and connected
  DigiKeyboard.delay(2000);
}
```
The program is uploaded without any problem, as usual, but absolutely nothing happens.
I'm on Ubuntu Linux (Pop_OS more precisely), the board appears well if I run ``lsusb``, and no errors with ``dmesg``.

SpenceKonde, 2021-02-18:
Yeah I could never make any of the VUSB ones work either. That's why they are not bundled with the core. When they are made to work, they will be included as part of the core (as I do on all my cores when the normal libraries don't work and I've made a version that does. 

My big problem was that because I could *NEVER* make *ANYTHING* work, I could not even determine whether the problem was computer configuration/drivers or bad compiler output, or - most likely both at the same time. I'm planning @ArminJo said he was going to take another look at whether this worked IIRC? 

I've been generally struggling to get enough time to do what needs to be done on the cores, as I keep having to go put out fires on DxCore/megaTinyCore (those both need another point release today or tomorrow, because the most recent release had a critical regression on megaTinyCore and key bugfix was missing on DxCore, and there was an idiotic problem with PWM on DxCore as well.

pcfreak1201, 2021-02-18:
Well, here is the output, when I try to compile/upload it:
```
Using board 'attinyx8micr' from platform in folder: /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1
Using core 'tiny' from platform in folder: /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1
Detecting libraries used...
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny88 -DF_CPU=16000000L -DCLOCK_SOURCE=18 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX8 -DARDUINO_ARCH_AVR -DHASUSB -DNEOPIXELPORT=PORTA -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/cores/tiny -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/variants/tinyX8_MH /tmp/arduino_build_165610/sketch/HID_t88.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for DigiKeyboard.h: [DigisparkKeyboard]
ResolveLibrary(DigiKeyboard.h)
  -> candidates: [DigisparkKeyboard]
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny88 -DF_CPU=16000000L -DCLOCK_SOURCE=18 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX8 -DARDUINO_ARCH_AVR -DHASUSB -DNEOPIXELPORT=PORTA -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/cores/tiny -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/variants/tinyX8_MH -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard /tmp/arduino_build_165610/sketch/HID_t88.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny88 -DF_CPU=16000000L -DCLOCK_SOURCE=18 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX8 -DARDUINO_ARCH_AVR -DHASUSB -DNEOPIXELPORT=PORTA -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/cores/tiny -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/variants/tinyX8_MH -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard/oddebug.c -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny88 -DF_CPU=16000000L -DCLOCK_SOURCE=18 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX8 -DARDUINO_ARCH_AVR -DHASUSB -DNEOPIXELPORT=PORTA -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/cores/tiny -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/variants/tinyX8_MH -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard/osccal.c -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny88 -DF_CPU=16000000L -DCLOCK_SOURCE=18 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX8 -DARDUINO_ARCH_AVR -DHASUSB -DNEOPIXELPORT=PORTA -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/cores/tiny -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/variants/tinyX8_MH -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard/usbdrv.c -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny88 -DF_CPU=16000000L -DCLOCK_SOURCE=18 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX8 -DARDUINO_ARCH_AVR -DHASUSB -DNEOPIXELPORT=PORTA -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/cores/tiny -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/variants/tinyX8_MH -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard/usbdrvasm.S -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=attiny88 -DF_CPU=16000000L -DCLOCK_SOURCE=18 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX8 -DARDUINO_ARCH_AVR -DHASUSB -DNEOPIXELPORT=PORTA -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/cores/tiny -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/variants/tinyX8_MH -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard /tmp/arduino_build_165610/sketch/HID_t88.ino.cp [... cut ...]
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-size -A /tmp/arduino_build_165610/HID_t88.ino.elf
Der Sketch verwendet 2710 Bytes (40%) des Programmspeicherplatzes. Das Maximum sind 6714 Bytes.
Globale Variablen verwenden 96 Bytes (18%) des dynamischen Speichers, 416 Bytes für lokale Variablen verbleiben. Das Maximum sind 512 Bytes.
```
And confirmed - board is listed with lsusb now, but nothing happens.
Because I don't find the fuse-settings, I read them with avrdudess: L:0xE1 H:0xDC E:0x06

prandeamus, 2021-02-18:
I tried some of this a while back with V-USB on MT-EH Tiny88 boards. I will see if I can dig out the source to a test app that did some basic MIDI rather than Keyboard, but the idea is the same. As I recall, the V-USB libraries assumed the pin mappings for VUSB on Digispark/Attiny85 hardware (B3, B4), but the pin numbers for MHET/Attiny88 are different. (D1, D2) ?

I can't swear to this but it might be a clue. I'll see if I can find the old code, though it was a test rather than anything that was production ready...

SpenceKonde, 2021-02-18:
But do *any* of them *work*? 

Also, are you aware that you are compiliing with two completely separate copies of the figispark vusb packages?

/home/stefan/.arduino15/packages/digistump/hardware/avr/1.7.1/libraries/DigisparkKeyboard
and 
/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkKeyboard 

These are of course different versions of the ile (dramatically), which explains why the one you were using with ATTinyCore didn't compile :-P 

But I never managed to make ANY working USB device with these except a mouse one which malfunctioned horrendously and left my computer in need of a reboot to restore mouse functionality with a different mouse.

SpenceKonde, 2021-02-18:
The progmem errors mean the code hasn't been revised since avr-gcc 4.3 was taken up by Arduino....

pcfreak1201, 2021-02-18:
Depending Digispark - yes I resently changed to https://github.com/ArminJo/DigistumpArduino from official ("old") Digispark-BSP
For me it's working with older digispark-stuff, but it's not a lot what I use. Normally, I don't use VUSB at all because of all the timing and compytibility problems. I have always a STM32-Bluepill-USBASP with true serial-port connected. That's all what I need ;-)

pcfreak1201, 2021-02-18:
> But do any of them work?

I checked my t85_CDC-Example: https://github.com/pcfreak1201/micronucleus_commandline/tree/main/CDC_Example
on the Digispark t85 -- and it still works with the core from ArminJo **AND** ATTinyCore!

```
Linking everything together...
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -Wall -Wextra -Os -Wl,--gc-sections -mmcu=attiny85 -o /tmp/arduino_build_463353/MicronucleusDigiCDC.ino.elf /tmp/arduino_build_463353/sketch/MicronucleusDigiCDC.ino.cpp.o /tmp/arduino_build_463353/libraries/DigisparkCDC/usbdrvasm.S.o /tmp/arduino_build_463353/libraries/DigisparkCDC/oddebug.c.o /tmp/arduino_build_463353/libraries/DigisparkCDC/osccal.c.o /tmp/arduino_build_463353/libraries/DigisparkCDC/usbdrv.c.o /tmp/arduino_build_463353/libraries/DigisparkCDC/DigiCDC.cpp.o /tmp/arduino_build_463353/core/core.a -L/tmp/arduino_build_463353 -lm
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_463353/MicronucleusDigiCDC.ino.elf /tmp/arduino_build_463353/MicronucleusDigiCDC.ino.eep
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_463353/MicronucleusDigiCDC.ino.elf /tmp/arduino_build_463353/MicronucleusDigiCDC.ino.hex
Bibliothek DigisparkCDC im Ordner: /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries/DigisparkCDC (legacy) wird verwendet
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size -A /tmp/arduino_build_463353/MicronucleusDigiCDC.ino.elf
Der Sketch verwendet 4512 Bytes (68%) des Programmspeicherplatzes. Das Maximum sind 6586 Bytes.
Globale Variablen verwenden 237 Bytes (46%) des dynamischen Speichers, 275 Bytes für lokale Variablen verbleiben. Das Maximum sind 512 Bytes.
/home/stefan/.arduino15/packages/ATTinyCore/tools/micronucleus/2.0a4/launcher -cdigispark --timeout 60 -Uflash:w:/tmp/arduino_build_463353/MicronucleusDigiCDC.ino.hex:i 
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.5
> Device signature: 0x1e930b 
> Available space for user applications: 6586 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 103  page size: 64
> Erase function sleep duration: 721ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
running: 100% complete
>> Micronucleus done. Thank you!
```
So there is nothing "broken" in your core.

SpenceKonde, 2021-02-18:
OMG YOU WERE ABLE TO GET A CDC DEVICE TO WORK? All I could ever get those
to do is generate errors trying to get the device descriptor.. ..

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

pcfreak1201, 2021-02-18:
LOL - nothing what I tuned to work :-)
But I tried to get it running on the MH-Tiny88, and it generates a device, but as you wrote, it can't generate a real device.
It only says: "hello". And the heartbeat also stops after some time. May be the chrystal - may be the tuning in usbdrvasm.S ... who knows.

prandeamus, 2021-02-19:
Not sure if this helps, but some ago (May 2020) I tried to create a simple MIDI controller using the USBMIDI library on top of VUSB for a MH ET Tiny88

The configurations settings are here, for the USBMIDI library and if you look around line 32 you'll see it assumes that VUSB is on D2,D3.
https://github.com/BlokasLabs/USBMIDI/blob/master/src/usbboard.h

So for this library I set these three #defines before including the MIDI library (and indirectly VUSB)
#define USB_CFG_IOPORTNAME      D
#define USB_CFG_DMINUS_BIT      2
#define USB_CFG_DPLUS_BIT       1

Since I think that on the MHET Live88 uses D1 for USB+ and D2 for USB-

I realise this for a library working at higher level of abstraction above VUSB but it seemed to work for me. Have not recompiled recently and your mileage may vary. If I get the time to retry it I will do so.

Hope this helps, but as the preceding comment says if it's impossible to get the timing right for this board maybe it's time to give up and get something with native usb support.

philibertc, 2021-03-08:
> So for this library I set these three #defines before including the MIDI library (and indirectly VUSB)
> #define USB_CFG_IOPORTNAME D
> #define USB_CFG_DMINUS_BIT 2
> #define USB_CFG_DPLUS_BIT 1

Okay, thanks, I will try it!

SpenceKonde, 2022-01-26:
There are no plans to systematically address this. I was given a sufficiently convincing explanation of why vUSB from the sketch sucks. I provide a pins_usb.h that lists which pins are connected to usb based on the menu selections. That's as far as I'm going to go on this tarpit.

closed 2022-01-26 by SpenceKonde

mqnc, 2022-09-30:
Sorry to revive this thing but did anyone make any progress? I also just tried to modify DigiCDC and run it on MH-Tiny88
(adding
```
#elif defined (__AVR_ATtiny88__)
#define USB_CFG_IOPORTNAME      D
#define USB_CFG_DMINUS_BIT      2
#define USB_CFG_DPLUS_BIT       1
```
)
but it didn't work. I sent the same code to a Digispark (ATTiny85, 16.5MHz) and it worked. What I find a bit curious is that the MH-Tiny88 board has a USB-plug on it and uses micronucleus, which in turn uses V-USB and works without problems. Here is the configuration:
https://github.com/micronucleus/micronucleus/blob/master/firmware/configuration/t88_default/bootloaderconfig.h
So I don't see any reason why this shouldn't work. My next step would be to litter the code with debug printf and compare between MH-Tiny88 and Digispark.

mqnc, 2022-09-30:
https://github.com/mqnc/MHTinyCDC