SpenceKonde/ATTinyCore #529: Board manager fails to install 1.3.3, 1.4.0, 1.4.1. and 1.5.0

SpenceKonde, 2021-03-02:
1.5.0 - Error downloading https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-i686-mingw32.tar.bz2
Nothing was installed. -> https://i.imgur.com/0cyzoM7.png

1.4.1 - CRC doesn't match. File is corrupted.
Nothing was installed. -> https://i.imgur.com/5iAIIna.png

1.4.0 - CRC doesn't match. File is corrupted.
Nothing was installed. -> https://i.imgur.com/1ylVJIu.png

1.3.3 - CRC doesn't match. File is corrupted.
Nothing was installed. -> https://i.imgur.com/UZBCgHO.png

1.3.2 - Installation completed.
This is the latest version that can be installed (at least for me)  -> https://i.imgur.com/M6occS6.png
But there is another issue with USBasp programmer, generic option selecion for "USBasp" is gone, only "USBasp (ATTinyCore)" can be selected which unfortunately does not work-> 
_The uploader returned an error
avrdude: Warning: cannot query manufacturer for device: No such file or directory
avrdude: Warning: cannot query product for device: No such file or directory
avrdude: error: could not find USB device with vid=0x16c0 pid=0x5dc vendor='www.fischl.de' product='USBasp'_

frustrating, sorry...

_Originally posted by @kratochviljan in https://github.com/SpenceKonde/ATTinyCore/discussions/528_

SpenceKonde, 2021-03-03:
None of these issues reproduce on 1.8.13. 

The old IDE version you are using reproduces your issue with 1.5.0. I get the same issue on 1.4.1 and 1.4.0 (I am baffled as to how you could have gotten anything else, inderstanding the nature of the issue) and do not get any error on 1.3.3  It looks like the outdated JRE packaged with the Arduino IDE does not recognize the CA (certification authority) who provides the SSL certificates for my website. This is highlights a rarely discussed dark side of SSL - you do need to have a magic list of trusted root CAs, and their certs, and that collection does need to be periodically updated. And applications that have dependencies on that often provide no mechanism for that other than updating the application. 

I can totally make it work on those newer versions (by mirroring the micronucleus binaries on an http server - though it is still going to install updated toolchain and avrdude versions which apparent, I forget exactly what it was that I wanted each of those for, but I remember being excited to switch to them because it solved some problems I'd been dealing with.

SpenceKonde, 2021-03-03:
http://drazzy.com/etc/package_httpdrazzy.com_index.json hosts a json file that doesn't refer to anything hosted on https://azduino.com (this will of course become more relevant in the near future when I move the json file itself to https, which has been a frequent request and is good security practice.

closed 2021-03-03 by SpenceKonde

farukmath, 2025-01-04:
I had the same problem.
there is a solution in the video below.

https://www.youtube.com/watch?v=sycSdI49hlY