SpenceKonde/ATTinyCore #533: fetched archive size differs from size specified in index

ArminJo, 2021-03-06:
Error during install: installing platform ATTinyCore:avr@1.5.1: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index

SpenceKonde, 2021-03-06:
Huh... wtf... wait, did I ever upload that even? 

No, apparently I did not.. .So the file it was downloading was a github 404 error page; should be fixed now. Thanks for reporting. Note that you may need to delete the (bad) copy of ATTinyCore-1.5.1.tar.bz2 from staging inside your Arduino15 folder, board manager makes strange caching decisions.

SpenceKonde, 2021-03-06:
Thanks for reporting this!

closed 2021-03-06 by SpenceKonde

mr-sneezy, 2021-06-17:
I think I've hit the sane library installation issue tonight.
I've been trying to update the Digistump library from the digistump.com version to this one.
I'm using Arduino 2.0.0-beta7
I get error:
-------------------------------
Tool arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 already installed
Downloading packages
digistump:micronucleus@2.5
Installing digistump:micronucleus@2.5
Failed to install platform: digistump:avr.
Error: 2 UNKNOWN: installing tool digistump:micronucleus@2.5: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index
------------------------------
I've uninstalled the old digistump.com library version and tried again with same result, also tried to install version 1.7.2 with same error result. 
Workarounds ?

mr-sneezy, 2021-06-17:
I tried the manual installation suggested in the Installation section of the readme. This resolved the above error, but on compilation I now get this error:

Compilation error: Error: 2 UNKNOWN: no FQBN provided

I do have Digistump selected as board type. I note the selection is grey not white in the board list for 'digispark'.
Any ideas welcome.

SpenceKonde, 2021-06-17:
That problem is coming from the Digistump board definitions - you are installing a different core, thiis is the ATTinyCore repo.

My core supports the digistump AVR boards (it supports every ATTiny with at least 2k of flash memory released before the 2016 revolution (post-2016 being handled by megaTinyCore) using no bootloader, 4k parts using optiboot or no bootloader, and the 8/16k parts using optiboot, no bootloader, or micronucleus (the digistump one) if there was ever a board manufactured that used it)  except the ATTiny26). My board package tries to download and install Micronucleus 2.5-azd1, no digistump in front.

mr-sneezy, 2021-06-18:
Hi mate.
Thanks for your patience with your reply (many coders get quite unkind when us lower lifeforms at times).
Looks like I incorrectly concluded I was using a board manager installer for a ATtiny85 MCU Development Board from this repo, my mistake, confused by all the error messages, issue threads, and 'digi' everything I guess. It was from ArminJo actually.

I have sorted out the issues tonight though, uninstalled the board manager and reinstalled an updated one from ArminJo's DigistumpArduino repo and now installs & then compiles fine (and can use EEPROM.h library fine in it, which was an earlier issue). 
Cooking with gas again here so to speak...
Thanks.