SpenceKonde/ATTinyCore #534: Can't upload by micronucleus in 1.5.1 - wrong path (Linux)

pcfreak1201, 2021-03-06:
Hi Spence,
after updating to 1.5.1 on my Linux, "I can't upload" the blink-sketch:
```
/home/stefan/.arduino15/packages/ATTinyCore/tools/micronucleus/2.5-azd1/micronucleus-2.5/micronucleus --no-ansi --run --timeout 60 /tmp/arduino_build_755075/Blink_Tiny88.ino.hex 
java.io.IOException: Cannot run program "/home/stefan/.arduino15/packages/ATTinyCore/tools/micronucleus/2.5-azd1/micronucleus-2.5/micronucleus": error=2, Datei oder Verzeichnis nicht gefunden (File or Folder not found)
```
The path is: /home/stefan/Arduino/arduino15/packages/ATTinyCore/tools/micronucleus/2.5-azd1/micronucleus
not: /home/stefan/.arduino15/packages/ATTinyCore/tools/micronucleus/2.5-azd1/**micronucleus-2.5**/micronucleus
(after copying the uploader in the required and generated path, it works )

SpenceKonde, 2021-03-06:
ARGH!!!!

On windows, works in 1.5.1 and not in 1.5.0 for the inverse reason.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

SpenceKonde, 2021-03-06:
1.5.1 is pulled, problem was actually with the windows tool archive, that will be updated, and a new release made available shortly.

SpenceKonde, 2021-03-12:
Thanks for reporting, 1.5.2 is now out, and it sounds like it works every where

closed 2021-03-12 by SpenceKonde