SpenceKonde/ATTinyCore #535: Missing pinout diagram for the new ATtinyX61 pinout style

MCUdude, 2021-03-10:
Come on, Spence, you need to notify me when you create a new pinout, so I can follow up with a new pinout diagram! In this case, I discovered that it was missing by accident.

Anyways, here it is! BTW there are also some other previously unreleased QFN pinouts you may find useful too 🙂 

[ATTiny_pinout.zip](https://github.com/SpenceKonde/ATTinyCore/files/6114157/ATTiny_pinout.zip)

SpenceKonde, 2021-03-11:
Woah - I forgot about the pin mapping diagram...  (and also forgotten I put that into 1.5.0 - thought I was saving that for 2.0.0 ) There's going to be a new one for the t1634 as well in ATTinyCore 2.0.0 (next version, so numbered because it may break some existing code) - I should never have released anything with the dumbass pin mapping I had for the t1634 (the correct pinout is IMO pretty obvious, straight clockwise from PA0 to PC2, so, pc4 is 15 and pc5 is 16, then reset gets to be pin 17 as one would expect)

2.0.0 is also going to need a couple of changes on other parts, will give you list when I know I have them all.

Thank you!

SpenceKonde, 2021-03-12:
Thaaanks!

closed 2021-03-12 by SpenceKonde