SpenceKonde/ATTinyCore #539: Fast PWM with Arduino, the inputs outputs don't work

ypatidisioannis, 2021-03-25:
Hello to the community!

I am using ATTINY85 and I am programming it through the AVRISP using an Arduino UNO.

This is the code I am using :

#include <avr/io.h> #include <avr/interrupt.h>

void setup() {

PORTB = 0;        //Reset values on port B

// After setting up the timer counter,
// set the direction of the ports to output
DDRB |= (1<<PB1) | (1<<PB0) | (1<<PB3); // Set the direction of PB1 to an output

// PLLCSR - PLL control and status register:
// PLL is a clock multiplier - multiplies system     8 MHz by 8 to 64 MHz
// PLL is enabled when:PLLE bit is enabled,
// CKSEL fuse is programmed to 0001.  This clock is
//   switched off in sleep modes!
PLLCSR |= (1<<PLLE);    // PLL enable

// Wait until the PLOCK bit is enabled
// before allowing the PCK to be enabled
//WaitForPLOCK();
//unsigned int i = 0;

while ((PLLCSR & (1<<PLOCK)) == 0x00)
{
    // Do nothing until plock bit is set
}

PLLCSR |= (1<<PCKE); // Enable asynchronous mode, sets PWM clock source

TCCR1 =
        (1<<CTC1)    | // Enable PWM
        (1<<PWM1A)   | // Set source to pck
        (0<<(CS13))  |
        (0<<(CS12))  |
        (1<<(CS11))  |
        (1<<(CS10))  | // Clear the pin when match with ocr1x
        (1<<COM1A1);
GTCCR =   (1<<PWM1B) | (1<<COM1B1);

// Set PWM TOP value - counter will count and reset
//  after reaching this value
//            OCR1C
// 100 kHz    159
// 80 kHz    199

OCR1C = 199;

// Enable Timer1 OVF interrupt
TIMSK = (1<<OCIE1A) | (1<<TOIE1);

sei();

OCR1A = 190;  //chaning th FR
}

void loop() 
{ 
PORTB |= (1 << PB3);
delay(500); PORTB &= ~(1 << PB3); delay(500);
 }

the problem is that when I use the PWM my rest of the outputs do not work . Any idea why is this happening?

SpenceKonde, 2021-04-18:
Uhm..... I do not see a Timer1 OVF ISR anywhere there? Yet you have that interrupt enabled. Without an ISR supplied, when it hits the timer1 overflow, it will jump to BADISR which jumps to reset without executing RETI (hey don't look at me, that's how avr-gcc handles it - there isn't really any better way available since there is no error reporting facility) it then proceeds to execute the sketch until hitting an sei(), at which point interrupts will be reenabled, and the (still set) overflow flag triggers the bad ISR again. 

Also, we're in Arduino Land, the init() function called before setup() has already enabled interrupts when setup() is called (which you should know - because millis() and delay() work in setup, and they both depend on interrupts being enabled)

SpenceKonde, 2021-04-18:
Oh, and even with an ISR provided. you rely on the millis()/delay() timekeeping, but this line breaks millis
```c++
TIMSK = (1<<OCIE1A) | (1<<TOIE1);
```
disables the millis timer interrupt, which is controlled by the TOIE0 bit of TIMSK.

SpenceKonde, 2021-04-25:
To clarify above 

> If this wasn't done, I would expect the PWM waveform to be wrong too
If *there wasn;t an init() function that enables interuprts before setup is called* I would expect the PWM waveform to be wrong due to timer1 getting reconfigured before the first time the timer0 overflows triggering the crash outlined above. 

All issues can be explained by bugs in user code. Closing as no response from originator and problems appear to be in user code.

closed 2021-04-29 by SpenceKonde