SpenceKonde/ATTinyCore #541: Document I2C buffer size

technoblogy, 2021-04-07:
Knowing the I2C buffer size is quite important for doing things like driving I2C OLED displays, and it's quite difficult getting it from the Wire library sources because there are so many of them, so I think it would be useful to have that stated in this section:

https://github.com/SpenceKonde/ATTinyCore#i2c-support

I believe it's 32 bytes on all chips. Is that correct?

SpenceKonde, 2021-06-01:
Beats the hell out of me. I haven't looked at that in years.

technoblogy, 2021-06-01:
We had a more involved discussion about this here:

https://github.com/SpenceKonde/megaTinyCore/issues/383

My conclusion was that I'd be happy with 32 bytes for everything; easier to remember.

SpenceKonde, 2021-06-01:
I think making it 32 across the board makes sense, unless there exist parts with only 128b of ram for which ATTinyCore's Wire,h will compile and fit in the flash.

SpenceKonde, 2022-02-03:
This is now in the docs for .2.0.0-dev branch

closed 2022-02-03 by SpenceKonde