SpenceKonde/ATTinyCore #542: Share same pins between SPI and I2C

chamroeunbm, 2021-04-10:
Hello,
I use attiny85. I can get my NRF24L01 to works and I2C LCD to works separately but when I merge them using this wiring below, it get hangs. Any suggestion?

[https://github.com/chaeplin/esp8266_and_arduino/tree/master/_51-attiny85-nrf24-lcd-getntp](url)

prandeamus, 2021-04-10:
Pull up resistors missing?

SpenceKonde, 2021-04-10:
I am shocked that anyone made that work....  I2C does not permit arbitrary non-I2C signals on SCL/SDA.

chamroeunbm, 2021-04-10:
> Pull up resistors missing?

I am not sure too but based on that link's diagram no resistors needed.

SpenceKonde, 2021-04-10:
**Pull up resistors are always required for I2C**. Sometimes they are included on the breakout board for I2C devices, sometimes not, sometimes they are included but you have to complete a solder bridge to enable them, etc.

chamroeunbm, 2021-04-10:
> I am shocked that anyone made that work.... I2C does not permit arbitrary non-I2C signals on SCL/SDA.

It look like the source code is from China. Anyhow, suggestion is much appreciated as I am new to all this.

SpenceKonde, 2021-04-10:
That all aside, the fact that you say it worked with just the LCD on the same pins would indicate that somehow, there are pullups there...

closed 2021-04-10 by SpenceKonde

chamroeunbm, 2021-04-10:
> That all aside, the fact that you say it worked with just the LCD on the same pins would indicate that somehow, there are pullups there...

LCD+SHT30 I2c(Humidity sensors) alone works. NRF24L alone also work. but when I put togethers, it's not.

SpenceKonde, 2021-04-10:
Closed because without more evidence that it is viable, I have no reason to expect that the configuration shown would work. 

multiple I2C devices on one bus is supported, I 2C prtotocol has mechanisms for devices to know when they;'re being talked to.