SpenceKonde/ATTinyCore #546: Burning Bootloader t85/Digispark by ISP doesn't work (avrdude error)

pcfreak1201, 2021-04-18:
Hi Spence,

I want to init a new bootloader with 4.3V BOD, but there is (still) a problem with parsing bootloader.bod_bits:

```
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x1e930b (probably t85)
avrdude: erasing chip
avrdude: reading input file "0xFE"
avrdude: writing efuse (1 bytes):

Fehler beim Brennen des Bootloaders. (= Error burning bootloader)
Writing | ################################################## | 100% 0.05s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xFE:
avrdude: load data efuse data from input file 0xFE:
avrdude: input file 0xFE contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of efuse verified
avrdude: reading input file "0b11011{bootloader.bod_bits}"
avrdude: invalid byte value (0b11011{bootloader.bod_bits}) specified for immediate mode
avrdude: read from file '0b11011{bootloader.bod_bits}' failed

avrdude done.  Thank you.
```

SpenceKonde, 2021-04-18:
First I've heard of this issue; you say still as if it had already been reported. If there is no open issue for it, I don't know there is a problem.

closed 2021-04-18 by SpenceKonde

pcfreak1201, 2021-04-18:
Sorry, wasn't an issue, only at discussions. Therefore it wasn' correct to write "still". Confimed is fixed - I wasn't able to see wrong value :-/