SpenceKonde/ATTinyCore #550: 841 16mhz internal No Bootloader Error EEPROM

bobzane, 2021-04-24:
When I burn the 16 MHz internal no bootloader version bootloader onto the Attiny 841 it says it worked, but when I try to upload a sketch it says no bootloader. I can still upload the >4v 8Mhz bootloader just fine and then sketches work. A blank sketch works but my sketch includes EEPROM and it doesn't work.

     ```sketch\MCP_lin.cpp.o (symbol from plugin): In function `safe_eeprom_write_byte(unsigned char*, unsigned char)':
     (.text+0x0): multiple definition of `safe_eeprom_write_byte(unsigned char*, unsigned char)'
     sketch\Lin_C8.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
     collect2.exe: error: ld returned 1 exit status
     exit status 1
     Error compiling for board ATtiny441/841 (No bootloader).```

bobzane, 2021-04-24:
I found the problem, I defined EEPROM twice so `safe_eeprom_write` function must not have a `#ifdefined` statement and the regular EEPROM does. So when I changed to the faster speed to see what the controller could do it broke it.

closed 2021-04-24 by bobzane

SpenceKonde, 2021-04-24:
Well, that *is* a bug in EEPROM.h - even with include guards, you can't define things like I tried to. 

But - what I'm going to try next is just not doing the "safe" slow-down for EEPROM writes, because a competent individual reported having done a bunch of flash writes with OSCCAL maxed without issue. That will be how it works in 2.0.0 (which will have a bunch of other sexy additions)