SpenceKonde/ATTinyCore #561: I don't think that RS485=xxINV actually works...

WestfW, 2021-05-15:
There is code in pin_defs.h to check for the B5_INV and other "inverted" bits, but the pin_def code needs to have those symbols actually defined.  For example, there is:

```
#define B0 0x200
#define B1 0x201
#define B2 0x202
```

I think this means that builds specified with _INV pin names will end up using the defaults instead.

I'll create a pull request soon...

SpenceKonde, 2021-05-15:
Huh, really? That was a custom request someone wanted for something... and like, they didn't complain, I thought they said it was working.....

WestfW, 2021-05-16:
It looks like it would work if the desired inverted pin was also the default...

Here, a3.lst is created with `make attiny1634 RS485=A3`, and a3_inv with  `make attiny1634 RS485=A3_INV`
Note that the a3_inv case does produce opposite sbi/cbi for the data pins, but ... not the right port or bit.

```
 diff -u a3.lst a3inv.lst 
--- a3.lst      2021-05-16 00:05:23.000000000 -0700
+++ a3inv.lst   2021-05-16 00:04:51.000000000 -0700
@@ -132,12 +132,11 @@
 
 #ifdef RS485
   RS485_DDR |= _BV(RS485);
-    3dc0:      83 9a           sbi     0x10, 3 ; 16
+    3dc0:      60 9a           sbi     0x0c, 0 ; 12
   #ifdef RS485_INVERT
   RS485_PORT |= _BV(RS485);
-  #else
+    3dc2:      68 9a           sbi     0x0d, 0 ; 13
   RS485_PORT &= ~_BV(RS485);
-    3dc2:      8b 98           cbi     0x11, 3 ; 17
   #endif
 #endif
 
@@ -622,9 +621,9 @@
       // put transceiver to output mode
       #ifdef RS485_INVERT
       RS485_PORT &= ~_BV(RS485);
+    3f76:      68 98           cbi     0x0d, 0 ; 13
       #else
       RS485_PORT |= _BV(RS485);
-    3f76:      8b 9a           sbi     0x11, 3 ; 17
       #endif
       // put char
       UART_UDR = ch;
@@ -637,9 +636,7 @@
       // put transceiver to input mode
       #ifdef RS485_INVERT
       RS485_PORT |= _BV(RS485);
-      #else
-      RS485_PORT &= ~_BV(RS485);
-    3f80:      8b 98           cbi     0x11, 3 ; 17
+    3f80:      68 9a           sbi     0x0d, 0 ; 13
     3f82:      08 95           ret
 ```

Here's the same diff, only with the patch I had in mind...

```
diff -u a3.lst a3inv.lst 
--- a3.lst      2021-05-16 00:10:53.000000000 -0700
+++ a3inv.lst   2021-05-16 00:11:09.000000000 -0700
@@ -135,9 +135,8 @@
     3dc0:      83 9a           sbi     0x10, 3 ; 16
   #ifdef RS485_INVERT
   RS485_PORT |= _BV(RS485);
-  #else
+    3dc2:      8b 9a           sbi     0x11, 3 ; 17
   RS485_PORT &= ~_BV(RS485);
-    3dc2:      8b 98           cbi     0x11, 3 ; 17
   #endif
 #endif
 
@@ -622,9 +621,9 @@
       // put transceiver to output mode
       #ifdef RS485_INVERT
       RS485_PORT &= ~_BV(RS485);
+    3f76:      8b 98           cbi     0x11, 3 ; 17
       #else
       RS485_PORT |= _BV(RS485);
-    3f76:      8b 9a           sbi     0x11, 3 ; 17
       #endif
       // put char
       UART_UDR = ch;
@@ -637,9 +636,7 @@
       // put transceiver to input mode
       #ifdef RS485_INVERT
       RS485_PORT |= _BV(RS485);
-      #else
-      RS485_PORT &= ~_BV(RS485);
-    3f80:      8b 98           cbi     0x11, 3 ; 17
+    3f80:      8b 9a           sbi     0x11, 3 ; 17
     3f82:      08 95           ret
 
 00003f84 <getch>:
```

closed 2021-05-18 by SpenceKonde