SpenceKonde/ATTinyCore #564: PCMSK0 not declared for Attiny4313 while it is declared for Attiny2313

hmeijdam, 2021-05-17:
I am running into an compilation error with PCMSK0.  See attached sketch i made for the Attiny2313,. It compiles and works as expected for the Attiny2313. But when I compile it for the Attiny4313 I get a compilation error 
 exit status 1
      'PCMSK0' was not declared in this scope

According the datasheet:
Comparison Between ATtiny2313A and ATtiny4313 ->     The ATtiny2313A and ATtiny4313 differ only in memory sizes. 

My test sketch is attached 

[test_Attiny2313_vs_Attiny4313.ino.txt](https://github.com/SpenceKonde/ATTinyCore/files/6496087/test_Attiny2313_vs_Attiny4313.ino.txt)

SpenceKonde, 2021-05-17:
Looking in  io headers. they named the PCMSK registers PCINT and PCINT1 instead of PCINT0 and PCINT1. 

I will add that to the list of stupid oversights in the x313 io headers that my core corrects in the next release. God they did a poor job with those parts! They started with crappy hardware, but then made the experience of working with them even worse with inconsistent headers

SpenceKonde, 2021-05-17:
```
#ifndef PCMSK0
#define PCMSK0 PCMSK
#endif 
```
put that at top of your sketch and it should work until the next version of ATTinyCore is available.

hmeijdam, 2021-05-17:
12 hours ago I thought it would be a walk in the park to port my code from Attiny13 to Attiny4313, I wanted to use the 16bit timer of the 2313/4313.  The datasheet indeed made me do some "reverse prayers".. but thanks to your suggested modification all works fine now. 
You can close the issue if you want, or leave it open until your next release.
Thanks,
Hans

SpenceKonde, 2021-05-17:
That's the only kind of prayer appropriate to the x313 parts. why they remain so widely used has always baffled me. 

Issue gets closed only when I commit the fix

hmeijdam, 2021-05-18:
If you are going to fix this in the next release: I compared the both header files "iotn2313a.h"  and "iotn4313.h" in directory 
X:\........\arduino-1.8.13\portable\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr

I saw that indeed the definition of PCMSK0 was missing in the 4313 file

But I also noted that the definition of RSIG  is missing for the 2313 header file, so for example  "  `SPMCSR |= _BV(RSIG); `  "
will only compile for the 4313 and not for the 2313

Maybe only a theoretical issue, as RSIG is the "Read Device Signature Imprint Table" bit, so who would ever use that in their code.

SpenceKonde, 2021-05-18:
Does that not work on my core? RSIG is missing from a lot of the classic AVR io headers and I thought I'd already put in a compatibility define for it. Maybe it's only in 2.0.0 branch or something.

RSIG is needed if you want to read the factory OSCCAL value, which becomes relevant when using an internal oscillator with tuning. I know I ran into the fact that it was missing from a bunch of parts because I needed it on the parts I was doing that on.

hmeijdam, 2021-05-18:
Try compiling this for Attiny2313 (works) or Attiny4313 (error) with your core.

`void setup() {
SPMCSR |= _BV(RSIG);
}

void loop() {
}`

SpenceKonde, 2021-05-18:
I'll take your word for it - I'm not going to go branch hopping on my development system (and right now in the 2.0.0 branch as I have it, nothing compiles) - I have uncommitted changes and it would be quite a hassle.

SpenceKonde, 2021-06-16:
Corrected for 2.0.0 by cc1723a455caf149b3fe944767c7e0b5cbfff1c0

closed 2021-06-16 by SpenceKonde