SpenceKonde/ATTinyCore #569: During installation via boards manager: >> Warning: non trusted contribution, skipping script execution

hmeijdam, 2021-05-23:
I just added Arduino IDE 1.8.15 to my list of portable installations and while I added all the different cores to it I noticed that when installing AttinyCore via the boards manager I get this warning

`Warning: non trusted contribution, skipping script execution (E:\Arduino\arduino-1.8.15\portable\packages\ATTinyCore\tools\micronucleus\2.5-azd1b\post_install.bat)
`
Could it have something to do with the fact that the boards manager URL http://drazzy.com/package_drazzy.com_index.json is:
http://   and not a   https://

SpenceKonde, 2021-05-23:
No, any code that is not signed by Arduino gets that warning. That check is being removed because someone (I suspect many people) pointed out that a malicious core can still execute arbitrary code with privileges of the person running it, just by telling the IDE that that malicious payload is a compiler.

The post_install.bat is left there so that as soon as we get versions of Arduino that remove the stupid check, it will work without me needing to rerelease, and to communicate to the user where the Micronucleus driver installer is. 

You can y

closed 2021-05-23 by SpenceKonde

hmeijdam, 2021-05-23:
Ok thanks for clarifying. Just to be sure I ran the post_install.bat file myself and it installed the drivers (that probably were already on my system). Not that I need the bootloader, but who knows in a few months and then I will have forgotten about the drivers completely.

SpenceKonde, 2021-05-24:
Yup
That's the driver used for boards like the Digispark and clones thereof - which can be programmed directly over USB. Despite the fact that the chip absolutely doens;'t have USB hardware support. Some individuals far more talented than I (or the digispark people - they just used the code changing it to polled instead of interrupt driven when being used to load code, VUSB has license terms that it's free as long as the code written from it is opensource) came up with a way to bitbang USB. As long as it's used in polled mode for programming only and we don't try to use USB from the Arduino sketch, it actually works surprisingly well considering what it is. Making anything USB work from the point of view of the sketch.... that's another story (note the issue about all USB libraries >.>) But used on it's own, or as part of a program written at a very low level  with keen awareness of that issue, it works quite well. Thats how USBAsps and USBTinyISP's work, too.