SpenceKonde/ATTinyCore #575: Attiny48 burn Bootloader with error

PixelBoy-tw, 2021-06-02:
Hi,I have burn Bootloader for attiny48 but it display error like this：

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\johnn\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/avrdude.conf"

         Using Port                    : COM26
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         Setting bit clk period        : 5.0
         AVR Part                      : ATtiny48
         Chip Erase delay              : 15000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no         64    4      0  3600  3600 0xff 0xff
           flash         65     6    64    0 yes      4096   64     64  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9209 (probably t48)
avrdude: erasing chip
avrdude: reading input file "0xfe"
avrdude: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xfe:
avrdude: load data efuse data from input file 0xfe:
avrdude: input file 0xfe contains 1 bytes
avrdude: reading on-chip efuse data:

燒錄bootloader時發生錯誤。
Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of efuse verified
avrdude: reading input file "0b11011111"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0b11011111:
avrdude: load data hfuse data from input file 0b11011111:
avrdude: input file 0b11011111 contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0xE2"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0xE2:
avrdude: load data lfuse data from input file 0xE2:
avrdude: input file 0xE2 contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified
avrdude: reading input file "C:\Users\johnn\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/optiboot/optiboot_attiny48_8000000L.hex"
avrdude: ERROR: address 0x2000 out of range at line 38 of C:\Users\johnn\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/optiboot/optiboot_attiny48_8000000L.hex
avrdude: read from file 'C:\Users\johnn\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/optiboot/optiboot_attiny48_8000000L.hex' failed

avrdude done.  Thank you.

SpenceKonde, 2021-06-08:
Goddamnit, I thought I responded to this over a week ago and was wondering why you hadn't reported back. 

Though i hadn't seen the cause at that time - though I do now, makefile is positioning the bootloader version number in the location it should be for an 88 instead of a 48. I don't know why it's compiling at all - that really ought to generate a compile error!... I wonder if because we're placing it directly, nothing ever checks that the address is sane.

PixelBoy-tw, 2021-06-09:
ok I see the problem,thx to respond !
I will close this issues

PixelBoy-tw, 2021-06-11:
Nope, I try it but it run error at line37

`Arduino:1.8.15 (Windows 10), 開發板:"ATtiny48/88 (Optiboot), Enabled, ATtiny48, 8 MHz (internal), Standard, B.O.D. Disabled (saves power), Enabled, Port A (pins 23~26)"

C:\Users\johnn\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\johnn\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/avrdude.conf -v -pattiny48 -cstk500v1 -PCOM30 -b19200 -e -Uefuse:w:0xfe:m -Uhfuse:w:0b11011111:m -Ulfuse:w:0xE2:m -Uflash:w:C:\Users\johnn\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/optiboot/optiboot_attiny48_8000000L.hex:i 

avrdude: Version 6.3-20201216

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\johnn\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/avrdude.conf"

         Using Port                    : COM30

         Using Programmer              : stk500v1

         Overriding Baud Rate          : 19200

         Setting bit clk period        : 5.0

         AVR Part                      : ATtiny48

         Chip Erase delay              : 15000 us

         PAGEL                         : PD7

         BS2                           : PC2

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         ByteDelay                     : 0

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                  Block Poll               Page                       Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom        65    20     4    0 no         64    4      0  3600  3600 0xff 0xff

           flash         65     6    64    0 yes      4096   64     64  4500  4500 0xff 0xff

           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500

         Description     : Atmel STK500 Version 1.x firmware

         Hardware Version: 2

         Firmware Version: 1.18

         Topcard         : Unknown

         Vtarget         : 0.0 V

         Varef           : 0.0 V

         Oscillator      : Off

         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9209 (probably t48)

avrdude: erasing chip

avrdude: reading input file "0xfe"

avrdude: writing efuse (1 bytes):

燒錄bootloader時發生錯誤。

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of efuse written

avrdude: verifying efuse memory against 0xfe:

avrdude: load data efuse data from input file 0xfe:

avrdude: input file 0xfe contains 1 bytes

avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...

avrdude: 1 bytes of efuse verified

avrdude: reading input file "0b11011111"

avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of hfuse written

avrdude: verifying hfuse memory against 0b11011111:

avrdude: load data hfuse data from input file 0b11011111:

avrdude: input file 0b11011111 contains 1 bytes

avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...

avrdude: 1 bytes of hfuse verified

avrdude: reading input file "0xE2"

avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lfuse written

avrdude: verifying lfuse memory against 0xE2:

avrdude: load data lfuse data from input file 0xE2:

avrdude: input file 0xE2 contains 1 bytes

avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...

avrdude: 1 bytes of lfuse verified

avrdude: reading input file "C:\Users\johnn\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/optiboot/optiboot_attiny48_8000000L.hex"

avrdude: ERROR: address 0x2000 out of range at line 37 of C:\Users\johnn\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/optiboot/optiboot_attiny48_8000000L.hex

avrdude: read from file 'C:\Users\johnn\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/optiboot/optiboot_attiny48_8000000L.hex' failed

avrdude done.  Thank you.

`

SpenceKonde, 2021-06-15:
This is frixzed in the 2.0.0-dev branch; note that there have been other changes to the bootloaders, and they are not compatible with the boards.txt in 1.5.x versions of the core, and the boards.txt in 2.0.0-dev branch is not updated either. At present, nothing should be expected to compile in the 2.0.0-dev branch I'vbe been checking things inb there as backupo, not because they are  done/usable.

SpenceKonde, 2021-06-15:
The upload speed in ATTinyCore for parts using soft UART was stupidly low, and I raised it so it wouldn't suck so much to upload to them 

(I think at the time I was being flat out vindictive, and was annoyed that [people wanted to use the boring parts without hardware serial - but still wanted the bootloader... so I kneecapped the bootloader I gave them. As it happens, soft UART on classic AVRs isn't much worse than hardware for the bootloader. It shits for the application, but that's beside the point. Classic AVR UART is kinda crappy. The fact that you get 2% baud rate inaccuracy for 115200 baud @ 16 MHz due to the granularity of UBRR is.... pretty gross for parts they were introducing in the first half of the 2010's. 

```
4800 below 4 MHz 19200 from 4 up to 8, 38400 from 8 to 12, and 56700 above that that when there's no hardware USART
```

Previously it was half that up to 8, and then 19200 all the way up. I don't think anything else in there was changed that would cause compatibility issues, though some other things were fixed I think.

commit c8157f06 (2021-06-15): Reorganize optiboot bootloader files

all binaries are now in hex subfolder and source in source subfolder. Since there are 170 hex files, we don't want them clogging up the directory with the source code.
Also fixed #575
closed 2021-06-15 by SpenceKonde