SpenceKonde/ATTinyCore #587: attiny ADC With internal 2.56v bug

mostafahk, 2021-06-25:
at analogRead() function following line is incorrect because REFS2 is in the ADCSRB register.

``` c
  #if defined(REFS2)
  ADMUX |= (((analog_reference & 0x04) >> 2) << REFS2); //some have an extra reference bit in a weird position.
  #endif
```

please change it to:
``` c
#if defined(REFS2)
ADCSRB |= (((analog_reference & 0x04) >> 2) << REFS2); //some have an extra reference bit in a weird position.
#endif
```

SpenceKonde, 2021-06-26:
On which part? The REFS bits beyond 0/1 are all over the place/

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

mostafahk, 2021-06-26:
I changed the function to the following code and my problem solved. REFS0/1 bits are in the ADMUX but REFS2 exists in the ADCSRB.

``` c
int analogRead(uint8_t pin)
{
  pin &=127; //strip off the high bit of the A# constants
  #ifndef ADCSRA
  return digitalRead(analogInputToDigitalPin(pin)) ? 1023 : 0; //No ADC, so read as a digital pin instead.
  #endif

  #if defined(ADMUX)
  ADMUX = ((analog_reference & ADMUX_REFS_MASK) << REFS0) | ((pin & ADMUX_MUX_MASK) << MUX0); //select the channel and reference
  #if defined(REFS2)
  ADCSRB |= (((analog_reference & 0x04) >> 2) << REFS2); //some have an extra reference bit in a weird position.
  #endif
  #endif

  #if defined(HAVE_ADC) && HAVE_ADC
  sbi(ADCSRA, ADSC); //Start conversion

  while(ADCSRA & (1<<ADSC)); //Wait for conversion to complete.

  uint8_t low = ADCL;
  uint8_t high = ADCH;
  return (high << 8) | low;
  #else
  return LOW;
  #endif
}
```

The current code doesn't work for the last two rows of voltage reference:
[image]

SpenceKonde, 2021-06-29:
REFS2 location is different depending on which part you are using, this core supports over a dozen parts. I ask you AGAIN

WHICH ATTINY PART DO I NEED TO CHANGE THIS FOR?

mostafahk, 2021-06-29:
ATTINY861

SpenceKonde, 2021-06-29:
Thank you! I will be sure that is corrected in 2.0.0!

closed 2022-01-28 by SpenceKonde