SpenceKonde/ATTinyCore #589: Invalid

NicoHood, 2021-07-09:
We are currently [facing an issue with Attiny x61 and PCINT](https://github.com/NicoHood/PinChangeInterrupt/issues/39#issuecomment-868266128). It turns out the `digitalPinToPCMSKbit` definition is wrong:

https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/variants/tinyX61_New/pins_arduino.h#L162

My wild guess is that this is correct:

```c
#define digitalPinToPCMSKbit(p) (((p) <= 7) ? (p) : ((p) - 7))
```

tredpath, 2021-07-09:
Verifying functionality on hardware. digitalPinToPCMSK before:
```
digitalPinToPCMSK(PIN_PA0) == &PCMSK0
digitalPinToPCMSK(PIN_PA1) == &PCMSK0
digitalPinToPCMSK(PIN_PA2) == &PCMSK0
digitalPinToPCMSK(PIN_PA3) == &PCMSK0
digitalPinToPCMSK(PIN_PA4) == &PCMSK0
digitalPinToPCMSK(PIN_PA5) == &PCMSK0
digitalPinToPCMSK(PIN_PA6) == &PCMSK0
digitalPinToPCMSK(PIN_PA7) == &PCMSK0
digitalPinToPCMSK(PIN_PB0) == &PCMSK0
digitalPinToPCMSK(PIN_PB1) == &PCMSK0
digitalPinToPCMSK(PIN_PB2) == &PCMSK0
digitalPinToPCMSK(PIN_PB3) == &PCMSK0
digitalPinToPCMSK(PIN_PB4) == &PCMSK0
digitalPinToPCMSK(PIN_PB5) == &PCMSK0
digitalPinToPCMSK(PIN_PB6) == &PCMSK0
digitalPinToPCMSK(PIN_PB7) == &PCMSK0
```

I assume the original was a typo so with digitalPinToPCMSK changed to `(((p) >= 0 && (p) <= 16) ? ((p<8)?(&PCMSK0) : (&PCMSK1)) : ((uint8_t *)NULL))`:
```
digitalPinToPCMSK(PIN_PA0) == &PCMSK0
digitalPinToPCMSK(PIN_PA1) == &PCMSK0
digitalPinToPCMSK(PIN_PA2) == &PCMSK0
digitalPinToPCMSK(PIN_PA3) == &PCMSK0
digitalPinToPCMSK(PIN_PA4) == &PCMSK0
digitalPinToPCMSK(PIN_PA5) == &PCMSK0
digitalPinToPCMSK(PIN_PA6) == &PCMSK0
digitalPinToPCMSK(PIN_PA7) == &PCMSK0
digitalPinToPCMSK(PIN_PB0) == &PCMSK1
digitalPinToPCMSK(PIN_PB1) == &PCMSK1
digitalPinToPCMSK(PIN_PB2) == &PCMSK1
digitalPinToPCMSK(PIN_PB3) == &PCMSK1
digitalPinToPCMSK(PIN_PB4) == &PCMSK1
digitalPinToPCMSK(PIN_PB5) == &PCMSK1
digitalPinToPCMSK(PIN_PB6) == &PCMSK1
digitalPinToPCMSK(PIN_PB7) == &PCMSK1
```

digitalPinToPCMSKbit before:
```
digitalPinToPCMSKbit(PIN_PA0) == 1
digitalPinToPCMSKbit(PIN_PA1) == 2
digitalPinToPCMSKbit(PIN_PA2) == 4
digitalPinToPCMSKbit(PIN_PA3) == 8
digitalPinToPCMSKbit(PIN_PA4) == 16
digitalPinToPCMSKbit(PIN_PA5) == 32
digitalPinToPCMSKbit(PIN_PA6) == 64
digitalPinToPCMSKbit(PIN_PA7) == 128
digitalPinToPCMSKbit(PIN_PB0) == 1
digitalPinToPCMSKbit(PIN_PB1) == 2
digitalPinToPCMSKbit(PIN_PB2) == 4
digitalPinToPCMSKbit(PIN_PB3) == 8
digitalPinToPCMSKbit(PIN_PB4) == 16
digitalPinToPCMSKbit(PIN_PB5) == 32
digitalPinToPCMSKbit(PIN_PB6) == 64
digitalPinToPCMSKbit(PIN_PB7) == 128
```

The original is creating a mask for the PCMSK value instead of just returning the bit index. Getting rid of the shift and using `((p) & 0x07)`:
```
digitalPinToPCMSKbit(PIN_PA0) == 0
digitalPinToPCMSKbit(PIN_PA1) == 1
digitalPinToPCMSKbit(PIN_PA2) == 2
digitalPinToPCMSKbit(PIN_PA3) == 3
digitalPinToPCMSKbit(PIN_PA4) == 4
digitalPinToPCMSKbit(PIN_PA5) == 5
digitalPinToPCMSKbit(PIN_PA6) == 6
digitalPinToPCMSKbit(PIN_PA7) == 7
digitalPinToPCMSKbit(PIN_PB0) == 0
digitalPinToPCMSKbit(PIN_PB1) == 1
digitalPinToPCMSKbit(PIN_PB2) == 2
digitalPinToPCMSKbit(PIN_PB3) == 3
digitalPinToPCMSKbit(PIN_PB4) == 4
digitalPinToPCMSKbit(PIN_PB5) == 5
digitalPinToPCMSKbit(PIN_PB6) == 6
digitalPinToPCMSKbit(PIN_PB7) == 7
```

I think digitalPinToPCICRbit is also incorrect, missing that PCIE1 includes PCINT15:12 as well as PCINT7:0 and PCIE0 only does PCINT11:8.
Before:
```
digitalPinToPCICRbit(PIN_PA0) == PCIE0
digitalPinToPCICRbit(PIN_PA1) == PCIE0
digitalPinToPCICRbit(PIN_PA2) == PCIE0
digitalPinToPCICRbit(PIN_PA3) == PCIE0
digitalPinToPCICRbit(PIN_PA4) == PCIE0
digitalPinToPCICRbit(PIN_PA5) == PCIE0
digitalPinToPCICRbit(PIN_PA6) == PCIE0
digitalPinToPCICRbit(PIN_PA7) == PCIE0
digitalPinToPCICRbit(PIN_PB0) == PCIE1
digitalPinToPCICRbit(PIN_PB1) == PCIE1
digitalPinToPCICRbit(PIN_PB2) == PCIE1
digitalPinToPCICRbit(PIN_PB3) == PCIE1
digitalPinToPCICRbit(PIN_PB4) == PCIE1
digitalPinToPCICRbit(PIN_PB5) == PCIE1
digitalPinToPCICRbit(PIN_PB6) == PCIE1
digitalPinToPCICRbit(PIN_PB7) == PCIE1
```

Using
```c++
digitalPinToPCICRbit(p) (((p) >= 8 && (p) <= 11) ? (PCIE0) : (PCIE1))
```
```
digitalPinToPCICRbit(PIN_PA0) == PCIE1
digitalPinToPCICRbit(PIN_PA1) == PCIE1
digitalPinToPCICRbit(PIN_PA2) == PCIE1
digitalPinToPCICRbit(PIN_PA3) == PCIE1
digitalPinToPCICRbit(PIN_PA4) == PCIE1
digitalPinToPCICRbit(PIN_PA5) == PCIE1
digitalPinToPCICRbit(PIN_PA6) == PCIE1
digitalPinToPCICRbit(PIN_PA7) == PCIE1
digitalPinToPCICRbit(PIN_PB0) == PCIE0
digitalPinToPCICRbit(PIN_PB1) == PCIE0
digitalPinToPCICRbit(PIN_PB2) == PCIE0
digitalPinToPCICRbit(PIN_PB3) == PCIE0
digitalPinToPCICRbit(PIN_PB4) == PCIE1
digitalPinToPCICRbit(PIN_PB5) == PCIE1
digitalPinToPCICRbit(PIN_PB6) == PCIE1
digitalPinToPCICRbit(PIN_PB7) == PCIE1
```

NicoHood, 2021-07-09:
@tredpath Thanks for verifying. What about creating a PR?

SpenceKonde, 2021-07-09:
Thanks for the PR. I will need to pull those changed over to the 2.0.0 branch so they get into the next branch (2.0.0 doesn;'t build anything currentlly)

SpenceKonde, 2022-02-02:
Is what I have in v2.0.0 dev correct?

SpenceKonde, 2022-02-03:
Confirmed that this change was carried over to 2.0.0 in full.

commit c74c0d0c (2022-01-06): Fix #589 correctly

Forgot the fix to the PCICR bit macro
closed 2022-02-03 by SpenceKonde