SpenceKonde/ATTinyCore #593: ATtiny841 micronucleus should not assume Nanite841 bootloader

bratoff, 2021-07-22:
Some 841 board variants will not function properly with the Nanite841 bootloader.  There should be either:

1. A menu in the existing board definition to choose Nanite841.hex or t841_default.hex
or
2. A new board definition "ATtiny841 micronucleus/generic" that uses t841_default.hex

Either of these can be accomplished by simple mods to boards.txt and by including t841_default.hex in the release.

SpenceKonde, 2021-07-27:
This is planned for 2.0.0. 

I discovered to my great dismay that various people making t84[1] boards must have been unaware of eachothers efforts or working to actively thwart one another.. The 84 and 841, between them, have had hardware produced with nearly every permutation of the pins on port B used for USB!

bratoff, 2021-07-27:
Fortunately my design works with the default micronucleus config for the t841 and uses the "ccw" pinout, so no problem there.

Do you have any minor releases planned prior to 2.0.0?  I could take a stab at a PR with just the bootloader change.

SpenceKonde, 2021-08-07:
No, 2.0.0 codebase has huge changes, internal changes amd I am strugging to find time to get that version through without trying to mangage an old release. Most likley to come soon is a partial 2.0.0 which won't have new versions of micronucleus but which peoplwe can at least use and PR against.

SpenceKonde, 2022-01-26:
New bootloaders are checked into v2.0.0-dev. There are a whopping 272 different binaries (136, plus upgrade for each of them). 48 of those binaries are for the 841, covering the three USB pinsets that have been used. Note that CCW vs CW doesn't matter for the bootloader, only for the sketch.. All of the recent work for 2.0.0 has been done in 2022, and it is nearing release 

 I just got the build script for micronucleus to work with the automatically generated configuration files on monday, so obviously the testing coverage has been ah, sparse. but public 2.0.0-dev testing has commenced, Closing this issue.

closed 2022-01-26 by SpenceKonde