SpenceKonde/ATTinyCore #594: Upload fails under Mac OS 10.13.6

CaptClaude, 2021-07-25:
This looks like a huge improvement over the previous library, and it _almost_ worked.
Is there a way around the error below for those of us not ready/unable to upgrade to Mac OS 10.15?
Or do I need to take this up with @ArminJo ??

>Please plug in the device (will time out in 60 seconds) ... 
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 1.6
> Available space for user applications: 6012 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 94  page size: 64
> Erase function sleep duration: 752ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
dyld: lazy symbol binding failed: Symbol not found: ____chkstk_darwin
  Referenced from: /Users/jm/Library/Arduino15/packages/ATTinyCore/tools/micronucleus/2.5-azd1b/micronucleus (**which was built for Mac OS X 10.15**)
  Expected in: /usr/lib/libSystem.B.dylib
dyld: Symbol not found: ____chkstk_darwin
  Referenced from: /Users/jm/Library/Arduino15/packages/ATTinyCore/tools/micronucleus/2.5-azd1b/micronucleus (which was built for Mac OS X 10.15)
An error occurred while uploading the sketch
  Expected in: /usr/lib/libSystem.B.dylib

SpenceKonde, 2021-07-25:
@stonehippo?
@cpldcpu?
@sanbeg?

I forget what went down to get the new version last winter.... (I can only address 64k of memory directly, just like the AVR architecture, so I can't keep track of tyhe details of what we did nback then - much less figure out how to work with these crazyass compilers for the supercomputers we all have on our desks. I also don't know anything about macs, other than that apple clearly knows that they can get away with pulling shit like this every OS upgrade, and take a sadistic joy in it. 

I remnember the shitstorm on the forums when apple broke the CH340G drivers i n catastrophic ways in what, 3 consecutive MacOS releases?

SpenceKonde, 2022-02-09:
No further reports and no response from maintainers of micronucleus.

closed 2022-02-09 by SpenceKonde

CaptClaude, 2022-02-10:
Bummer. Thanks for trying.