SpenceKonde/ATTinyCore #597: feature suggestion - Simulator option for readers

matititam, 2021-08-13:
Dear Team

First of all, thanks a lot for the library which is well maintained. 
I am writing this to ask your opinion on the Arduino simulator which can be used by the readers of this libarry to try out a lot of examples before htey go ahead and integrate it in their projects. 

here is one link:

SpenceKonde, 2021-09-25:
There isn't exactly a team. It's me, and occasionally I've managed to convince people to help me with things I had no idea how to do.  And I maintain 2 other cores, have a damned day job,, and sell on Tindie.... which is why releases have beeen so slow over here, I need to get the C++17 changes we decided on over there into those two,, do a DxCore release some other fixes, then check in the new wire (which is in unreleased megaTinyCore already), I'm planning to come back and finish 2.0.0 here, during which time people will be banging on and finding bugs in the new wire library over there. When I finally get 2.0.0 out, it's going to be a pretty big deal, since it both cleans up the godawful mess surrounding the 

busy-ness is also why I had't realized I hadn't responded to this. 

Anyway 

I write embedded C and AVR assembly. I don't have a clue how to do something like that, it looks like it would require a massive effort to create anything like that... Or are you saying that this simulator exists and there is some way that my core could be made to support it? If that's the case, what needs to happen? 

I am always happy to accept PR's that improve documentation, add examples, fix bugs (though right now, due to the pending 2.0.0 release. this isn't the best timeto go bug hunting; Tons of bugs were fixed and things changed that aren't yet in release or master, only 2.0.0-dev)  though if you've got anything big planned, we should probably make sure that it's aligned with the changes coming in 2.0.0 (the new pins_arduino.h files are done and located in 2.0.0-dev branch now. Nothing builds there though, because before I could finish it something blew up back over on megaTinyCore.

Again, really sorry for missing this when you created it.

SpenceKonde, 2022-02-07:
2.0.0 draws near and is available for public testing. Still very curious about what you were suggesting with this issue (and 0% chance of it happening without more explanation)

matititam, 2022-02-07:
Dear @SpenceKonde 

I will be happy to test the new 2.0.0 core on the [Arduino simulator](https://wokwi.com). Kindly keep me posted. I would like to share more working examples of Arduino projects using ATTinyCore with your readers.

Thanks

SpenceKonde, 2022-08-14:
Adding blocked because I have no idea how to do this