SpenceKonde/ATTinyCore #600: tinyNeoPixel: ColorHSV and gamma32 unimplemented

stonehippo, 2021-08-18:
I wanted to use `ColorHSV()` and `gamma32()` with the tinyNeoPixel/tinyNeoPixel_Static libraries.

I thought they were available (they show up in the library header), but when trying to use them, I got an ld error about an undefined reference, which lead me to peek into the implementation, and sure enough, they're not there.

I assume these were defined for compatibility with the Adafruit_NeoPixel library. Any particular reason they're not implemented? If not, would a PR to include them be welcome?

stonehippo, 2021-08-18:
I see that they've been implemented in megaTinyCore. Maybe I can just port those over.

SpenceKonde, 2021-08-18:
Yeah I just hadn't gotten around to poetyongvthem to here it was on list
for the long delayed 2.
0.0

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

stonehippo, 2021-08-18:
Got it (and thanks for all the work you do on this and megaTinyCore!). If I can find time in the next couple of days, I'll shoot you a PR.

SpenceKonde, 2021-08-18:
thanks, - the branch to pr to is 2.0.0-dev, 
NOT master. (note that nothing compiles over there due to lack of boards.txt - that'sthe next phase.

closed 2022-02-02 by SpenceKonde