SpenceKonde/ATTinyCore #617: Wire.readBytes

TheFidax, 2021-11-01:
Is possible add readBytes function in Wire library?
Code tested and working correctly with Arduino Core (UNO, Nano), Mini Core(ATmega328), Mega Core(ATmega128), with ATtiny841 has this error:

`error: 'class TwoWire' has no member named 'readBytes'`

SpenceKonde, 2021-11-05:
stream looks like it needs to be updated. (Wire inherits this from Stream)

TheFidax, 2021-11-19:
In 2.0.0 release, is possible add another function not implemented:  Wire.setWireTimeout() ?
Documentations are available here: https://github.com/arduino/Arduino/issues/10803

SpenceKonde, 2021-11-19:
All parts except the 441, 841, and 828 will get that i 2,.0.0 when https://github.com/SpenceKonde/ATTinyCore/pull/618 is merged in. I2C master on these parts is hideous kludge becuse there;sno support whatsioever for I2C master in hardware.

SpenceKonde, 2022-01-27:
Actually, it turrns out that #618 predates extensive changes to the wire library and I cannot merge it into 2.0.0. Sorry.

SpenceKonde, 2022-02-07:
Status: Wire.readBytes should work in 2.0.0-dev, Wire.setTimeout requires attention from an expert to implement in the 2.0.0 codebase since #618 could not be merged. Closing this issue as the title issue is solved and the second issue is not something I intend to take on. I already have refactored that monstrosity twice for 2.0.0-dev. (which is why the PR which was submitted against the dead 1.5.x branch instead of the current development branch couldn't be merged)

closed 2022-02-07 by SpenceKonde