SpenceKonde/ATTinyCore #620: Does this library work with Attiny85-20mur?

urkku, 2021-11-12:
Does this library work with Attiny85-20mur? I need to use The pwm singnal on PB3 and the pin mapping image that this library is saying that there is no pwm on PB3. 

The datasheet of Attiny85-20mur clearly states that PB3 is Pwm pin.  

https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-2586-AVR-8-bit-Microcontroller-ATtiny25-ATtiny45-ATtiny85_Datasheet-Summary.pdf Link

sleemanj, 2021-11-12:
The short answer to your question is no, you can not perform PWM on that pin.

The long answer is that PB3 (over-barred OC1B) is a complementary (inverted) output of PB4, it is not a "PWM" pin in Arduino speak, the chip can only drive it if both PB3 and it's non-inverted PB4 is driven simultaneously (same signal, inverted), see Table 12-2 on page 87 of the full Tiny85 datasheet, it can not be driven on it's own or independantly, so it's not something you would find useful as a "PWM" pin in Arduino speak.

urkku, 2021-11-12:
Thank you for fast and precise answer. I did not even realize that.

closed 2021-11-12 by urkku

SpenceKonde, 2021-11-12:
Yup. I looked into this ages ago - there just isn't a way to bully a third PWM output out of that timer (EVEN THOUGH IT HAS THREE OCR registers! The third one just sets period :-( , and the inverted outputs are basically useless for PWM as we know it..

SpenceKonde, 2021-11-12:
(as an aside - I thought about but never implemented support for sacrificing some precious flash on the 412 to get 3 PWM channels from the TCA (which can, when doing 3 or less pins, be made to do 16-bit PWM instead of 8) , and 2 from the TCD for PWM on all available I/O pins; the problem was just that having both versions of analogWrite took up more flash than I was happy with, and all the 412 users are really concerned about flash because they never made an 8k one. The 20-pin parts are, as it happens, available in a 3mm x 3mm QFN package, but good luck soldering that!

prandeamus, 2021-11-12:
As another aside there's a discussion and explanation from David Johnson-Davies at http://www.technoblogy.com/show?LE0 "Four PWM Outputs from the ATtiny85" with ways to "fake" 4 PWM outputs with some interrupt handlers. Demo program to control an RGBW LED.

prandeamus, 2021-11-12:
And another aside, sorry ... 

"and the inverted outputs are basically useless for PWM as we know it."

I think there is one use case where you can have a loudspeaker connected to the complementary PWM to get more power in a push-pull configuration. Or so I think. It's 37 years since I studied this lark.

urkku, 2021-11-15:
Thank you all for the proper explanation. 
I did try the David Johnson 3PWM fake approach but didnt get it to work properly:  https://medium.com/jungletronics/fast-rainbow-circuit-f0dddd5ee28