SpenceKonde/ATTinyCore #623: Where is tiny841?

Linzm99, 2021-11-28:
I cant find the attiny841 in the board selector did I install it incorrectly? Where should it be normally?

hmeijdam, 2021-11-28:
[image]

Is what you are looking for?

Linzm99, 2021-11-28:
Thanks for the fast reply. Turns out I installed the tinymegacore and got cunfused...

SpenceKonde, 2021-11-29:
ATTinyCore = anything released before 2016. The most significant digits of the part number indicate flash size, and is always a power of 2. The rest of it identifies which family/series it is, which determines the features, number of pins, and the pinout. The numbers are not systematically assigned, and one family doesn't tell you anything about the pin assignments of another. 

megaTinyCore = anything released after 2016 - these will have a 2, 4, 6, or 7 in the 1's place (indicating the pincount - 8, 14, 20, or 24 pin), and a 0, 1, or 2 in the tens place (indicating the featureset - 1-series has the type D timer, 3 analog comparators, a DAC, and two 10-bit single-ended ADCs, and 2 CCL logic blocks. 2-series has 1 analog comparator, no type D timer or DAC, but 4 logic blocks, a second USART, and a *VERY FANCY* 12-bit ADC, and the 0-series is a lobotomized 1-series that's a few cents cheaper). Within a series, all parts (regardless of pin count) have the same features, and the same functions are on the same pins (except for 8-pin parts, probably why the 2-series doesn't have 8-pin parts) and moving between series, no functionality that is present is on a different pin; they add and remove features, but they don't shuffle the pin assignments around like classic tinys would. 

The two cores present a similar interface, but share essentially no code. megaTinyCore shares most of it's code with the DxCore code, which supports the AVR Dx-series, which supports the latest-and-greatest full size parts in the DA, DB, and future DD-series.

Microchip is going to have a problem when they get to the next tinyAVR series, because a "3-series" tinyAVR with 16k of flash and 14 pins would be "ATtiny1634" - but that's already taken by a classic tinyAVR part..... Of course, that's assuming they even continue the ATtiny branding. The DD-series is encroaching onto tinyAVR territory - there's going to be an AVR64DD14 with 14 pins, and the series goes down to 16k flash sizes, and it they've already killed "ATmega" - they released the "0-series" ATmegas back in 2016ish timeframe, but what you'd have expected to be the 1-series was released with DA/DB branding.

closed 2021-11-29 by SpenceKonde