SpenceKonde/ATTinyCore #626: Use only serial TX and recycle the RX pin on ATTINY85

imayoda, 2021-12-06:
Hi there, and thank you for excellent core,
As per subject, is there the possibility to spare PB1 on an ATTINY85 in Serial operation? 
I only need the TX pin and the RX pin would be nice to have for other tasks.

Any suggestion would be much appreciated, Thanks in advance :)

SpenceKonde, 2021-12-22:
Thisis planned for 2.0.0. Currently you can manually achieve it (albeit without the flash savings by. after Serial.begin() writing 
`ACSR=0` or `ACSRA=` (depending on the part.

SpenceKonde, 2022-01-19:
Man, I did some serious work on that software serial. That code is OLD or the author wasn't particularly adept at assembly. I get to do a similar round of optimization on pulseInASM too (and that benefits all three cores with smaller code size)

On TX I borrowed the trick that neopixel library uses of precalculating the two patterns for PORT for a 1 and a 0, since the pin can change we can't rely on it being converted to a sbi/cbi. But since the PORT is known, we can use OUT not ST, since the classic tinies dont have 1 cycle ST (which sucks)

And fully integrated the tricks with the carry bit (stage the character with 0x80, clear carry at the beginning, the uart-delay routine doesn't mess with the carry bit since it uses dec not subtraction, and then for each bit, we sample the pin with SBIC (skip if bit in io register is clear) and the thing we skip is SEC (SEt Carry)... and then we follow that with a ROR, (Roll Over Right through carry) to shift in the new bit, and loads the former low bit into carry. Then we check if carry is clear. If it's not, then we've gotten the 8 bits (the first was the 0x80 we started with. At that point we can return to user code as soon as we know we've hit the stop bit,. Easy peasy. I may have to tweak the baud calculation due to the improved performance :-P 

Of course I can't test it because it's all in 2.0,0, and that doesn't have a working boards.txt yet. Which is what I ought to be doing.

imayoda, 2022-01-20:
Sorry for the delay, but, whoa, it literally outstands even my ability to understand the gimmicks of software serial on this tiny chips.. :)
Meanwhile I worked it out by switching off the serial with end() after every serial write (and resetting RX pin to output).. it works as I don't need extreme code execution speed. 
Thank you for every effort you put in this cores!

SpenceKonde, 2022-01-20:
Problem with that method is you still use as much flash as you would if you had RX capability..

SpenceKonde, 2022-01-26:
This change is available for testing in v2.0,0-dev. Along with a few other improvements. It is significantly more efficient now. Baud rate accuracy needs to be verified - I may have imporoived it enough that the timing was changed.

imayoda, 2022-01-29:
I'm testing asap when I put my hands again on the now live project :)
Meanwhile thanks a lot

SpenceKonde, 2022-02-02:
This appears to work.

closed 2022-02-02 by SpenceKonde