SpenceKonde/ATTinyCore #628: attiny85 USBasp burn bootloader verification error

Heheovereggs, 2021-12-15:
I install library through additional board manager URL,
tool->board I select is **Attiny45/85 (Optiboot)**, and the programmer is **USBasp (AttinyCore)**
The hardware I have is a Chinese USBasp purchased from amazon and attiny85 is brand-new from Digikey.
Tried on both 2.0.0-rc2 and 1.8.18 versions of Arduino IDE

Error output:
`avrdude: set SCK frequency to 187500 Hz
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: set SCK frequency to 187500 Hz
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
 ***failed;  
avrdude: verification error, first mismatch at byte 0x0000
         0x00 != 0xdf
avrdude: verification error; content mismatch
Error while burning the bootloader: Failed chip erase: uploading error: exit status 1
`

Please let me know if there is any additional information I can provide

SpenceKonde, 2021-12-15:
Recheck the connections very carefully. Remember that MISO and MOSI pins for ISP programming are swapped from what thy are when using the ATTiny as master to control an SPI slave. MISO goes to tiny85's DO pins MOSI goes to DI pin - the ISP programming uses the same hardware as the USI which a lot of people (rightly so) find very confusing. 

"failed chip erase"? Makes me suspect bad firmware on the USBasp - there are like a million versions of the FW that ship on USBAsps, and not all of them are good

Try with another Arduino and Arduino as ISP -at least that way, you know whthat the code is in theory possibly able to 
it's goddwful, but at least it rules out a USBAsp with bad firmware.

pcfreak1201, 2021-12-15:
Had also these messages some times with my cheap chinese USBasp. I could solder a jumper to "slow", so the firmware should reduce the SCK for 1MHz clock on ATTINY85, if the jumper is set. It happens always with new chips, never when already set to 8MHz or 16.5MHz with a sketch ...

SpenceKonde, 2021-12-15:
oh god arethey still shipping UASBAsps that don't autoadjust frequencty!!!?

What you want is Bill Perry's firmware for it, which fixes a shitlaod of obnoxious bugs

pcfreak1201, 2021-12-15:
Chinese sellers newer dump parts in the junk what are looking good. They are only using accounts with no reputation :-)
You mean https://github.com/bperrybap/usbasp/tree/master/bin/firmware ?
I switched to a Bluepill with this firmware: https://github.com/amitesh-singh/FASTUSBasp

bperrybap, 2021-12-19:
You want the 1.06alpha branch.
Yeah, I guess I need to do some cleanup on the repository.

closed 2021-12-20 by SpenceKonde