SpenceKonde/ATTinyCore #629: Sleep mode on arrony1614 but not a issue.

Therealjosephchrzempiec, 2021-12-16:
Hello this is not a issue but a question. I’m trying to get attiny1614. I’m trying to put it into sleep mode and wake it up every 10 seconds. However I can’t not find any example code online that can do that. Is there some place that I can’t ale a look at? I’m not good at coding I need some help please.

Joseph

SpenceKonde, 2021-12-16:
1/ Youi are in the wrtong core. the ATtny1614 is not supported by ATTinyCore, it is supported by megaTinyCore. 
Have you read the crude powersave documentation linked to grtom the main readme? it is really basic and written by spomeoen else because I have never used sleep or powersave on these parts - that libraruy has been something I really ewant to do do and is the #1 priority for both DxCore and megaTinyCore (it will be3 one library for both as their sleep functionality is identical) That is all the documentation I have available.
 Currently my list of things that I will be spending my time on is:
Forced move and not gettign fired, at least not before the end of year bonus goes out.
Release DxCore 1.4.0 amd megaTinyCore2.5.0 (tonight - these are MASSIVE updates with TONS of imprvovements. 
megaTinyCore changelog:
```
* **New I2C/TWI implementation** (I2C and TWI refer to the same interface, the one provided by Wire.h; don't get me started about how these and the library are named).
  * Support for acting as both master and slave (on the same pins); this configuration, sometimes known as "multi-master", includes not only the simple case of multiple masters and slaves on an I2C bus, each of which is always exclusively either a master or a slave, but also the more complicated case, which has been a frequent request: For the AVR device to act as both master AND slave. It can both initiate transactions with slaves, or respond to transactions initiated by other masters.
  * New tools menu option to select whether to support only being a master *or* a slave, as we do now (default) or to support acting as both master *and* slave (new functionality).
  * Support for Dual Mode (one instance of TWI acting as master on one pair of pins and slave on another) on parts that support it (there are no current or announced tinyAVR parts with dual mode support.
  * Significantly reduced flash usage under all circumstances (even master+slave mode should use less flash than old master or slave mode - the cost of supporting master-and-slave mode is RAM for a second buffer. There is an implementation included that can use the same memory for both buffers, however, it is not currently exposed via an option due to the risk of breakage if you receive while preparing to send something.
  * Failed attempts will timeout instead of hanging indefinitely.
  * Support for slave counting the bytes read by the master, and for slave to check whether it's in the middle of a transaction (for example, before sleeping)
* Recent change to C++17 required additions to new.cpp and new.h, including sized deallocation (`delete`) and alignment-aware `new` and `delete` operators. The sized deallocation operator is called when existing code that worked before is compiled to the C++ 17 standard; since free() doesn't care about the size, implementation was straightforward. Discussion is ongoing about the aligned `new` and `delete` operators, which are also new in this version of the standards. It is likely that we will not support them, since other Arduino cores aren't even building to C++ 17 standard, so if your code needs aligned new/delete, it also won't work anywhere else in Arduino-land. While we are not shy about adding features, we do so only to support hardware features. If conditions change we will revisit this matter.
* Using millis or micros (or rather attempting to) when they are unavailable due to millis being disabled, or in the case of micros, RTC used as millis time source, give better errors.
* Clarified licence (for one thing, renamed to a .md so people can read it more easily, and know that it's readable if they're on windows) for tinyNeoPixel.
* Improved docs for tinyNeoPixel. The two libraries each have a README.md linking to a greatly expanded library document.
* Document use of WDT for it's original purpose, to protect against hangs!
* Actually prevent disabling warnings - -Wall for all! You should not be compiling code with warnings disabled, the vast majority of the time, they're pointing to problems, and in the cases that aren't bugs, they're still a weak point that future bugs could come from (and that people will comment on when you post the code on forums to ask for help). I thought I'd done this a long time ago. Also pull in some warning-related flags from DxCore, including making implicit function declarations error, since the implied declarations when this happens are basically never result in working code, but it occasionally compiles and malfunctions.
* Fix timekeeping on clock speeds only supported with external clocks or tuning when a TCA or TCB is used for millis (it's still busted with the TCD)
* Correct SYSCFG0 fuse settings when burning bootloader for 2-series parts - they default the reserved bits to 1 not 0, and worse still, setting them to 0 enables a mode we probably don't want.
* Stop clearing fuse 4 by writing the default values for TCD0 on a 1-series. Now, with great difficulty, we only set that on parts that actually have the type D timer in order to keep our promise of burn bootloader restoring the chip to a fully known state. (well, except for the user row, and EEPROM if you've got it set to retain).
* Fix theoretical EEPROM.h bug inherited from avr-libc, and keep millis() from losing time when writing more than one byte at a time; update and harmonize with DxCore.
* Harmonize Comparator.h with DxCore.
* Fix 402 with bad signature support.
* Fix names of .lst and .map
[... cut ...]  the bad decisions imposed by the digispark team for he digiispark pro and copied by the clones)
A new basic 1634 breakout boards is also unarguably needed  (and a t841 one) because 2.0.0 will officiailly deprecate the pin numberts printed on the boards I sell. which isn't a good look. 

So that is the stuff that is coming ahead of the major project to bring sleep and powersave to the tinyAVR and DX-series.....
Sleep mode and powersaving library and demonstration code for megaTinyCore and DxCore

Therealjosephchrzempiec, 2021-12-16:
I'm so sorry. I thought it was in this one. My fault.

Joseph

SpenceKonde, 2021-12-16:
Whelp, you've now gotten a full outline of the order opf operationf fopr the next few months of development of these cores!

Therealjosephchrzempiec, 2021-12-16:
Thank you, I'm reading it all.

Joseph

SpenceKonde, 2021-12-16:
That bit about not buying from someone on the grounds that the hardware will have flaws based om oter poor decisions you've seen them make. saved me a few hundred dollars last year.  I almost bought a bunch of serial adapters that did some very vvisible things right.... but the routing looked like the person was really inexperienced, and left a bad taste in my mouth. Since then I've found 4 design defects, 2 of them criticall and all of which would have forced me to puit them straight into the trashcan.

Therealjosephchrzempiec, 2021-12-17:
I love the new attiny processors. 

Joseph

closed 2021-12-20 by SpenceKonde