SpenceKonde/ATTinyCore #635: Installation affects toolset used for other AVR platforms.

bperrybap, 2021-12-19:
This is an issue in the IDE / builder tool but perhaps there is a work around in this platform.
I don't understand the mechanism at play but when this package is installed using the boards manager in the IDE, it affects which gcc toolset is used by other AVR platforms.
I have seen this issue with another platform - the MegaAVR platform from arduino.cc

I have MANY IDE versions (pretty much all of them going back to before 1.0) and many IDE core platforms installed on a linux machine. 
Each IDE version comes bundled with its own toolset that is used for builds when using that version of the IDE.
But that changes once this platform is installed.
For example,
before this package is installed, a build for an UNO board will use the gcc tools down under the directory for that IDE.
But after this platform is installed, the gcc tool set used switches to use the one down under the packages directory.
On my system it switches to:
~/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin
instead of using the tools that came with the IDE.

This is problematic for many reasons.

I had a discussion about it with the Arduino.cc developers and they don't see this as an issue and see what they are doing as a feature. i.e. why wouldn't you always want to use later / newer tools?
But it is a problem since it now becomes possible to have a core or library code that is shipped with an IDE using an incompatible toolset.
Tools that may be incompatible not only with the source code in the bundled core and libraries but with with 3rd party libraries, user sketches,  boards, and bootloaders.
Also, by using alternate tools,  it bypasses any modifications that may have been made to the source code or tools in the IDE sub directories that were specific to that IDE bundle.
Overall, overriding the IDE bundled toolsets is a very bad thing. Afterall, there were many reasons the tools were finally bundled with the IDE releases about 10 years ago.

Great example a while back. In the early MegaAVR platform release, they broke lots and lots of code that worked with the bundled IDE tools and core. It broke so bad that  some existing code for UNO boards wouldn't even compile after that platform was installed.
This is because not only can they decide to use newer toolsets, but under certain conditions, they can also decide to pull core code or libraries from a newer "compatible" core/platform.

I'm not sure what causes it to suddenly switch to using a different toolset after this platform is installed, but I also noticed that if I download a zip of the repository and install it manually under the sketchbook/hardware directory, or clone the repo into that area this does not happen.

SpenceKonde, 2021-12-20:
This should only effect the "stock" arduino platform, and only if it hasn't been updated through board manager. The real problem is that there is no equivalent to the entry in the board manager json file for the default arduino AVR package that the IDE ships with to specify the version.

And so all manually installed cores, as well as an unupgraded default core, use the most recent version of the toolset available. That may not be the one that the IDE shipped with.

Any core that supports any vaguely recent processor needs to specify a new version of the toolchain to ensure that it will work on versions of the IDE that shipped with older versions of the toolchain. If you are modifying the toolchain, you must not use board manager. 

This is why I have half a dozen installations of the Arduino IDE, some "portable" and others which I never use board manager on for development. 

Closing, this is not something that I can fix without breaking the core, the fact that board manager installations can impact other cores is a bug in the Arduino IDE.

closed 2021-12-20 by SpenceKonde

bperrybap, 2021-12-20:
Yeah, the IDE is broken by design in this area. For the life of me I can't figure out why the Arduino.cc team can't see it.

This relates to their wrong headed decision many years ago to decision to never update anything inside the IDE installation area.
The issue was one of permissions, particularly in in a multi user environment, since the IDE could be installed in a way that users could not update it. (which could be argued is a good thing) But there were other ways to solve that that didn't create all the secondary issues due to using/sharing user updates in common areas outside of the IDE installation area in the users home directory. 
It was just a really bad decision. It can create all kinds of problems since there is no way to ever truely lock down a given IDE bundle.
It even breaks the ability to use package manages on linux.
The IDE even foolishly puts bundled library updates for a given IDE into the users sketchbook/libraries directory which is shared by all IDEs installed on the machine.

I'm guessing that they will likely eventually move to snap or flatpak, which are easier for developers and solves some of these issues, but both those tools can create all kinds of security issues for users since depending on how the developer uses them, the developer can override system shared libraries with their own private versions effectively making the application statically linked.