SpenceKonde/ATTinyCore #644: MOSI and MISO Pins to be swapped in attinyx4

pprakash, 2022-01-14:
Am working on RF24 communication between ATTiny84 and UNO. During the process I noticed that the MOSI and MISO pins are swapped in the pins_arduino.h file. 
`#define MOSI 5 
#define MISO 6 `

As per the pin out [https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x4.png](url) MOSI should be at 6 and MISO should be at 5. 

I tried swapping this in the pins_arduino.h file but its not taking effect. Please let me know the procedure for swapping the same.

SpenceKonde, 2022-01-14:
I have renamed this topic to accurately reflect the nature of this issue. This is a documentation problem, bug in the behavior of the core. 

I have no idea what you hoped to achieve by changing pins_arduino.h - it can't change which pin has what function. The pins that have those functions are set by the hardware and cannot be changed.

pprakash, 2022-01-14:
@SpenceKonde ... thank you very much for the clarification, Now its clear on why the pins are changing in Master Mode when compared to ISP mode. Will have my connections swapped.

sleemanj, 2022-01-14:
On my simple pinouts I show both, "Programmer" (yellow) and "SPI" (pink) orientations for clarity - https://github.com/sleemanj/ArduinoOrientedChipPinoutCreator/blob/master/x4.png

SpenceKonde, 2022-01-14:
Isn't that the same thing I do?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

sleemanj, 2022-01-14:
[Your pinout image](https://raw.githubusercontent.com/SpenceKonde/ATTinyCore/master/avr/extras/ATtiny_x4.png) has master mode SPI pins denoted as DI and DO, only the programming (ISP) are shown as MOSI and MISO.

Certainly if I was looking at your diagram for a quick "now which pins are the ones to use for SPI" I would see MOSI and MISO and say "oh it's those pins are the ones to use for SPI" :-)

sleemanj, 2022-01-14:
I'd do something like this

[image]

or like this

[image]