SpenceKonde/ATTinyCore #656: Where do you want the V2 bugs to be posted?

hmeijdam, 2022-01-28:
Where do you want the V2 bugs to be posted?

e.g. this one with the "constrain" command.

On the "master" version below code compiles fine, but in the V2 dev it throws an error.
```
void setup() {
  int sensVal = constrain(sensVal, 10, 150);
}

void loop() {
}
```

results in error when compiling on V2.0.0-dev

```
Arduino: 1.8.13 (Windows 10), Board: "ATtiny85/45/25 (No Bootloader), ATtiny85, 8 MHz (internal), EEPROM retained, RX on PB1, TX on PB (default PB0), B.O.D. Disabled (saves power), Enabled"

E:\Arduino\arduino-1.8.13-attinycore\arduino-builder -dump-prefs -logger=machine -hardware E:\Arduino\arduino-1.8.13-attinycore\hardware -hardware E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-attinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-attinycore\hardware\tools\avr -built-in-libraries E:\Arduino\arduino-1.8.13-attinycore\libraries -libraries E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\libraries -fqbn=ATTinyCore:avr:attinyx5:chip=85,clock=internal_8m,eesave=aenable,softserial=enable,bod=disable,millis=enabled -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_596433 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_11383 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_491786\sketch_jan26a.ino

E:\Arduino\arduino-1.8.13-attinycore\arduino-builder -compile -logger=machine -hardware E:\Arduino\arduino-1.8.13-attinycore\hardware -hardware E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-attinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-attinycore\hardware\tools\avr -built-in-libraries E:\Arduino\arduino-1.8.13-attinycore\libraries -libraries E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\libraries -fqbn=ATTinyCore:avr:attinyx5:chip=85,clock=internal_8m,eesave=aenable,softserial=enable,bod=disable,millis=enabled -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_596433 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_11383 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_491786\sketch_jan26a.ino

Using board 'attinyx5' from platform in folder: E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr

Using core 'tiny' from platform in folder: E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr

Detecting libraries used...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_596433\\sketch\\sketch_jan26a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Generating function prototypes...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_596433\\sketch\\sketch_jan26a.ino.cpp" -o "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_596433\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE

"E:\\Arduino\\arduino-1.8.13-attinycore\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_596433\\preproc\\ctags_target_for_gcc_minus_e.cpp"

Compiling sketch...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny85 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_596433\\sketch\\sketch_jan26a.ino.cpp" -o "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_596433\\sketch\\sketch_jan26a.ino.cpp.o"

In file included from C:\Users\Hans\AppData\Local\Temp\arduino_build_596433\sketch\sketch_jan26a.ino.cpp:1:0:

C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_491786\sketch_jan26a.ino: In function 'void setup()':

E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny/Arduino.h:54:5: error: 'typeof' was not declared in this scope

     typeof (x) _x = (x);             \

     ^

C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_491786\sketch_jan26a.ino:2:17: note: in expansion of macro 'constrain'

   int sensVal = constrain(sensVal, 10, 150);

                 ^~~~~~~~~

E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny/Arduino.h:54:5: note: suggested alternative: 'feof'

     typeof (x) _x = (x);             \

     ^

C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_491786\sketch_jan26a.ino:2:17: note: in expansion of macro 'constrain'

   int sensVal = constrain(sensVal, 10, 150);

                 ^~~~~~~~~

E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny/Arduino.h:57:3: error: '_x' was not declared in this scope

   _x < _l ? _l : _x > _h ? _h :_x;})

   ^

C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_491786\sketch_jan26a.ino:2:17: note: in expansion of macro 'constrain'

   int sensVal = constrain(sensVal, 10, 150);

                 ^~~~~~~~~

E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny/Arduino.h:57:8: error: '_l' was not declared in this scope

   _x < _l ? _l : _x > _h ? _h :_x;})

        ^

C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_491786\sketch_jan26a.ino:2:17: note: in expansion of macro 'constrain'

   int sensVal = constrain(sensVal, 10, 150);

                 ^~~~~~~~~

E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny/Arduino.h:57:23: error: '_h' was not declared in this scope

   _x < _l ? _l : _x > _h ? _h :_x;})

                       ^

C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_491786\sketch_jan26a.ino:2:17: note: in expansion of macro 'constrain'

   int sensVal = constrain(sensVal, 10, 150);

                 ^~~~~~~~~

E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny/Arduino.h:57:36: error: void value not ignored as it ought to be

   _x < _l ? _l : _x > _h ? _h :_x;})

                                    ^

C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_491786\sketch_jan26a.ino:2:17: note: in expansion of macro 'constrain'

   int sensVal = constrain(sensVal, 10, 150);

                 ^~~~~~~~~

exit status 1

Error compiling for board ATtiny85/45/25 (No Bootloader).

```

_Originally posted by @hmeijdam in https://github.com/SpenceKonde/ATTinyCore/discussions/648#discussioncomment-2049725_

SpenceKonde, 2022-01-28:
New issues are good. Maybe prefix them with [V2]

But I think I fixed this one this morning... this is still happening? It's caused by typeof() not working when compiled with std=c++17, instead of gnu++17, but platform.txt is now set to use gnu++17

hmeijdam, 2022-01-28:
Are you sure you checked that in? When I do a fetch, the last change I get is 15hrs old. The delete issue in  gnu++17

[image]

SpenceKonde, 2022-01-28:
It was in the optimization menu checkin. That is apparently not the bug I thought it was. 

Constrain is just busted and I don't think it ever worked here. they call  a variable L in one place and low in another...

hmeijdam, 2022-01-28:
> 
> Constrain is just busted and I don't think it ever worked here. they call a variable L in one place and low in another...

So you are saying it was pure luck it worked in the Master release? Or at least compiles there without problem.

hmeijdam, 2022-01-28:
The definition of "constrain" in arduino.h in V2.0.0-dev is:
```
#ifndef constrain
  #define constrain(x,low,high)   ({ \
    typeof (x) _x = (x);             \
    typeof (low) _l = (l);           \
    typeof (high) _h = (h);          \
  _x < _l ? _l : _x > _h ? _h :_x;})
#endif
```
The definition of "constrain" in arduino.h in Master is:

`#define constrain(amt,low,high) ((amt)<(low)?(low):((amt)>(high)?(high):(amt)))
`

hmeijdam, 2022-01-28:
If I do a redefine of constrain in V2.0.0-dev according the "old" definition it compiles, just with a warning that I am doing a redefine.
(110 bytes in V2.0.0-dev and 78 bytes in Master)

```
#define constrain(amt,low,high) ((amt)<(low)?(low):((amt)>(high)?(high):(amt)))

void setup() {
  int sensVal = constrain(sensVal, 10, 150);
}

void loop() {
}
```

SpenceKonde, 2022-01-28:
Oh god the releases were still using THAT version of constrain? That version is no good. I didn't realize I had modified those macros
I have already fixed this on my system, with the correct version of constrain. The old version is no good, consider constrain(foo++, min, max)  - what value will foo have after constrain?  That's why it got replaced (thought I did that far longer ago that the past year or so.

closed 2022-02-02 by SpenceKonde

hmeijdam, 2022-02-02:
Confirmed, though some size increase

in Master  Sketch uses 248 bytes

in 2.0.0-dev  Sketch uses 280 bytes (-Os) 

and I get a valid warning (as it makes little sense to constrain a variable without a defined value).

```
In file included from E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny/Arduino.h:8:0,
                 from C:\Users\Hans\AppData\Local\Temp\arduino_build_209233\sketch\sketch_feb02a.ino.cpp:1:
C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_316738\sketch_feb02a.ino: In function 'setup':
E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny/BasicMacros.h:106:21: warning: 'sensVal' is used uninitialized in this function [-Wuninitialized]
   __typeof__ (x)    _x = (x);           \
                     ^~
```

This throws no warning
```
int sensVal = 100;

void setup() {
  sensVal = constrain(sensVal, 10, 150);
}

void loop() {
}
```