SpenceKonde/ATTinyCore #657: [V2] Using "Serial" while selecting "-O3" (speed) compiler flag, throws linker error

hmeijdam, 2022-01-29:
When using a "serial.**** command compilation fails if selecting "-O3" (speed) flag. ".
-Os" (size) and "-Og" (debugging) flags work fine.

Test sketch:
```
void setup(){
  Serial.begin(9600);
}

void loop() {
  Serial.print(" errors with -O3 flag ");
}
```

Error message:
```
Arduino: 1.8.13 (Windows 10), Board: "ATtiny85/45/25 (No Bootloader), -O3 (speed), ATtiny85, 8 MHz (internal), EEPROM retained, RX on PB1, TX on PB (default PB0), B.O.D. Disabled (saves power), Enabled"

E:\Arduino\arduino-1.8.13-attinycore\arduino-builder -dump-prefs -logger=machine -hardware E:\Arduino\arduino-1.8.13-attinycore\hardware -hardware E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-attinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-attinycore\hardware\tools\avr -built-in-libraries E:\Arduino\arduino-1.8.13-attinycore\libraries -libraries E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\libraries -fqbn=ATTinyCore:avr:attinyx5:optimization=othree,chip=85,clock=internal_8m,eesave=aenable,softserial=enable,bod=disable,millis=enabled -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_515615 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_327545 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_153719\TemperatureRead.ino

E:\Arduino\arduino-1.8.13-attinycore\arduino-builder -compile -logger=machine -hardware E:\Arduino\arduino-1.8.13-attinycore\hardware -hardware E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-attinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-attinycore\hardware\tools\avr -built-in-libraries E:\Arduino\arduino-1.8.13-attinycore\libraries -libraries E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\libraries -fqbn=ATTinyCore:avr:attinyx5:optimization=othree,chip=85,clock=internal_8m,eesave=aenable,softserial=enable,bod=disable,millis=enabled -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_515615 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_327545 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_153719\TemperatureRead.ino

Using board 'attinyx5' from platform in folder: E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr

Using core 'tiny' from platform in folder: E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr

Detecting libraries used...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -O3 -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_515615\\sketch\\TemperatureRead.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Generating function prototypes...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -O3 -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_515615\\sketch\\TemperatureRead.ino.cpp" -o "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_515615\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE

"E:\\Arduino\\arduino-1.8.13-attinycore\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_515615\\preproc\\ctags_target_for_gcc_minus_e.cpp"

Compiling sketch...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -O3 -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny85 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_515615\\sketch\\TemperatureRead.ino.cpp" -o "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_515615\\sketch\\TemperatureRead.ino.cpp.o"

Compiling libraries...

Compiling core...

Using precompiled core: C:\Users\Hans\AppData\Local\Temp\arduino_cache_327545\core\core_83044d6b95b50e776e4c6052de14a59a.a

Linking everything together...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-gcc" -Wall -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mrelax -mmcu=attiny85 -o "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_515615/TemperatureRead.ino.elf" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_515615\\sketch\\TemperatureRead.ino.cpp.o" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_515615/..\\arduino_cache_327545\\core\\core_83044d6b95b50e776e4c6052de14a59a.a" "-LC:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_515615" -lm

C:\Users\Hans\AppData\Local\Temp\ccOf2wr9.s: Assembler messages:

C:\Users\Hans\AppData\Local\Temp\ccOf2wr9.s:309: Error: symbol `_txstart' is already defined

C:\Users\Hans\AppData\Local\Temp\ccOf2wr9.s:312: Error: symbol `_txpart2' is already defined

C:\Users\Hans\AppData\Local\Temp\ccOf2wr9.s:314: Error: symbol `_txdelay' is already defined

C:\Users\Hans\AppData\Local\Temp\ccOf2wr9.s:710: Error: symbol `_txstart' is already defined

C:\Users\Hans\AppData\Local\Temp\ccOf2wr9.s:713: Error: symbol `_txpart2' is already defined

C:\Users\Hans\AppData\Local\Temp\ccOf2wr9.s:715: Error: symbol `_txdelay' is already defined

lto-wrapper.exe: fatal error: E:\Arduino\arduino-1.8.13-attinycore\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

e:/arduino/arduino-1.8.13-attinycore/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board ATtiny85/45/25 (No Bootloader).

```

SpenceKonde, 2022-02-02:
Fixed with the other fixes to tinySoftSerial to make it transmit and receive serial data, it now compiles with -O3

SpenceKonde, 2022-02-02:
fix checked into V2.0.0-dev

closed 2022-02-02 by SpenceKonde

hmeijdam, 2022-02-02:
Confirmed.
-Os = 834 bytes
-O3 = 978 bytes
-Og = 1150 bytes