SpenceKonde/ATTinyCore #672: Compiler

JohnOH, 2022-02-13:
I am building for the T84 and am seeing an error that I don't know how to progress.

/var/folders/6c/gxr4fz9s0_1cfcfnq7b8fmr40000gp/T//cce28EUt.s: Assembler messages:
/var/folders/6c/gxr4fz9s0_1cfcfnq7b8fmr40000gp/T//cce28EUt.s:2483: Error: register r24, r26, r28 or r30 required
lto-wrapper: fatal error: /private/var/folders/6c/gxr4fz9s0_1cfcfnq7b8fmr40000gp/T/AppTranslocation/303FD067-0E2F-4A92-B2BE-36EFB06A7CC8/d/Arduino-1-8-13.app/Contents/Java/hardware/tools/avr/bin/avr-gcc returned 1 exit status
compilation terminated.
/private/var/folders/6c/gxr4fz9s0_1cfcfnq7b8fmr40000gp/T/AppTranslocation/303FD067-0E2F-4A92-B2BE-36EFB06A7CC8/d/Arduino-1-8-13.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ATtiny84/44/24 (No Bootloader).

If someone could point me to some source I could take a look at I would be grateful.

SpenceKonde, 2022-02-13:
There is a problem with a piece of inline assembly. 
Looks like an instruction that only works on a "Special Upper Register Pair" (r24 (which implies r24 and r25), or the X, Y or Z pointer registers) is being told to use a register that isn't one of those. That indicates a defect in a piece of inline assembly; check your constraints: There's only two instructions that work only on the special upper register pairs, sbiw and adiw yet you're trying to use one of them with a variable not specified with the 'w', 'e', 'x', 'y' or 'z' constraint. The compiler is dutifully following your constraints as it substitutes in available registers, and in this case it is picking a register that isn't valid. This may be within code that doesn;t generate any errors in other contexts, because it depends on what registers are free at the poiint where the inline assembly block is called. 

You can get the assembly that the compiler is passing to the compiler through approximately the process I described over here:
https://github.com/SpenceKonde/AVR-Guidance/blob/master/LowLevel/SyntaxError_InlineASM.md - somewhere (somewhere after line 2483 - as noted you get gibberish out if you don;t gun the LTO, and hence the assembly generated becomes larger) you'll likely find an adiw or sbiw instruction called on something other than one of those 4 registers. Track that back to the assembly that created it and fix your constraints.

JohnOH, 2022-02-13:
You are very good, the code is below:

     volatile("sbiw      %0, 0x01 \n\t"
                 "ldi %1, 0xFF \n\t"
                 "cpi %A0, 0xFF \n\t"
                 "cpc %B0, %1 \n\t"
                 "brne .-10 \n\t"
                 : "+r" (delay), "+a" (tmp)
                 : "0" (delay)
                 );

SpenceKonde, 2022-02-13:
> The asm is in my code. The code has worked fine for a number of years with
> earlier versions of Arduino/cores. I need to do some research to understand
> the constraint terminology. Thank you for debugging my code.

Yeah. It was always wrong. Its just that in the past you must have always
called i5 where it wasn't already using all of the call used registers from
24 up.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2022-02-16:
As an aside, that assembly is strange...

1. Why is the LDI in the loop? 
2. specifying delay as both input and output operand with the "+" constraint modifier is redundant. 
3. If you need a scratch register (I don't think you actually do), why not just use __temp_reg__ (r0) 
4. You're making it decrement the counter by 1, until the counter is -1. Why would you not simply pass a value 1 larger, and test if it's reached zero - that doesn't need any additional instructions since the SREG zero flag tested by brne is set when SBIW results in 0.  comparison is just subtraction that sets the flags in SREG but otherwise throws away the result of the subtraction rather than storing it... 
The classical minimum size delay loop for > 256 iterations is just 4 clocks
```
"sbiw %[delay], 1 \n\t"
"brne .-4             \n\t"
:[delay] "+w" ((uint16_t) delay)  
);
```
I'd rather pad my asm with nop's than with unnecessary code if you need slightly longer delays than you can fit in a uint16_t.... 
A delay loop that took a 32-bit delay value wouldn't be particularly onerous or slow - unless I'm mistaken, this 6 clock loop can do delays of up to 4.2 billion times 6. .
```
"sbiw %0, 1 \n\t" //subtract 1 from low two bytes
"subc %0C, __zero_reg__ \n\t"
"subc %0D,__zero_reg__ \n\t" // Now you know why avr-gcc dedicates a register to storing a known zero!
"brne .-8"                         \n\t
:"+w" {{uint32_t) delay));
```

closed 2022-02-20 by SpenceKonde