SpenceKonde/ATTinyCore #673: MH attiny88 having default blinking LED but cant make my PC recognize it what ever i do!!

eadh21, 2022-02-21:
hi everyone, sorry if this isnt new but i didn't find any solutions yet.
i have the mh tiny attiny88 board, which has the blinking LED program as default. when i connect it via USB (data cable) to my pc it cant recognize it. it recognizes all other boards like esp32, esp 01 , Arduino uno.
i added the link in preferences but the PORT doesn't appear.

SpenceKonde, 2022-02-21:
Have you also downloaded and installed the drivers?

Nothing is expected to show in ports menu, those devices are dugispark
like, they are uploaded to using that tool, direct via USB. No serial port
appears or is involved.
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

eadh21, 2022-02-21:
yes i successfully installed the drivers.

i didnt get the last part, do u mean i can upload programs using arduino IDE without choosing the port ?

if u recommend another IDE ill try it out

SpenceKonde, 2022-02-21:
Yeah, select the micronucleus board definition and it will upload over USB
wand nothing that the computer sees as a serial port is involved.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

eadh21, 2022-02-21:
thanks for the fast response, i did as u said but it stuck at plug in the device, i plug it in many times after hitting compile but no luck

[image]

i found tutorial online where they can see it as COM port and it upload successfully after plug in

[image]

eadh21, 2022-02-21:
### **PROBLEM SOLVED**

i reinstalled bootloader micronucleus with zadig , as i understand it this program created a driver for my device and let PC recognize it.

for anyone who has the same problem (PC dont recognize MH-ET LIVE ATTINY88 and dont make sound when plugin)
use official drivers first if no luck try a program called zadig with the configuration for micronucleus.

[image]

i could not upload them here

closed 2022-02-22 by eadh21