SpenceKonde/ATTinyCore #681: Document printf() support

technoblogy, 2022-03-23:
I was pleased to discover that ATTinyCore supports **printf()**, like megaTinyCore and DxCore, so it would be nice to mention it in the documentation.

Also, perhaps there should be a **printf()** option on the **Boards** menus to allow you to choose **Default**, **Full**, or **Minimal** like the other cores?

SpenceKonde, 2022-03-27:
I will make sure the existence of this feature is documented, but are you sure it really isn't mentioned somewhere? The docs are not as well organized here as they are on megaTinyCore and DxCore (and this undertaking on DxCore was weeks of work).

technoblogy, 2022-03-27:
Thanks for the reply. Searching on the home page for **printf** doesn't find anything - I haven't looked deeper.

commit fe184f2e (2022-04-17): Fix #681
closed 2022-04-18 by SpenceKonde