SpenceKonde/ATTinyCore #688: AtTinycore and WDT

storfusker, 2022-04-12:
Hi there

I’ve spend the last 3 evenings trying to figure out why the watchdog on my AtTiny841 did not work properly
After many hours of searching, I read how to get a .LST file from the AtTinycore compilation.
Big applause to the person who implemented that very useful function!! 

My problem was that sometimes the watchdog behaved strange, and sometimes it did not work at all when I left out the wdt_reset()t in the loop

Original code with bad watchdog setup / function

void setup() {
  MCUSR &= ~(1<<WDRF);
 312:	84 b7       	in	r24, 0x34	; 52
 314:	87 7f       	andi	r24, 0xF7	; 247
 316:	84 bf       	out	0x34, r24	; 52
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:51
  CCP = 0xD8;
 318:	88 ed       	ldi	r24, 0xD8	; 216
 31a:	80 93 71 00 	sts	0x0071, r24	; 0x800071 <__DATA_REGION_ORIGIN__+0x11>
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:52
  WDTCSR &= ((0<<WDIE) | (0<<WDE));
 31e:	81 b5       	in	r24, 0x21	; 33
 320:	11 bc       	out	0x21, r1	; 33

As can be seen,  the WDTCSR setting doesn’t work so no wonder that the watchdog is not functioning properly.
The WDTCSR register, with correct address 0x21, is read into r24, and just written back without any modifications at all !!
Maybe the problem is the shift of “0” to position WDIE & WDE to set both to “O” to disable them both
Modifying the code to use negative logic give the same faulty result. WDTCSR is just read into r24 and immediately written back

void setup() {
  MCUSR &= ~(1<<WDRF);
 312:	84 b7       	in	r24, 0x34	; 52
 314:	87 7f       	andi	r24, 0xF7	; 247
 316:	84 bf       	out	0x34, r24	; 52
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:51
  CCP = 0xD8;
 318:	88 ed       	ldi	r24, 0xD8	; 216
 31a:	80 93 71 00 	sts	0x0071, r24	; 0x800071 <__DATA_REGION_ORIGIN__+0x11>
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:52
  WDTCSR &= (~(1<<WDIE) | ~(1<<WDE));
 31e:	81 b5       	in	r24, 0x21	; 33
 320:	81 bd       	out	0x21, r24	; 33

And using a different notification does not make any difference. WDTCSR is just read into r24 and immediately written back again

C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:52
  WDTCSR = WDTCSR & (~(1<<WDIE) | ~(1<<WDE));
 31e:	81 b5       	in	r24, 0x21	; 33
 320:	81 bd       	out	0x21, r24	; 33

Splitting the WDTCSR  setting in 2 lines instead makes it work. The register WDTCSR is now properly AND’ed and written back with correct value

void setup() {
  MCUSR &= ~(1<<WDRF);
 312:	84 b7       	in	r24, 0x34	; 52
 314:	87 7f       	andi	r24, 0xF7	; 247
 316:	84 bf       	out	0x34, r24	; 52
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:51
  WDTCSR &= ~(1<<WDIE);
 318:	81 b5       	in	r24, 0x21	; 33
 31a:	8f 7b       	andi	r24, 0xBF	; 191
 31c:	81 bd       	out	0x21, r24	; 33
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:52
  CCP = 0xD8;
 31e:	88 ed       	ldi	r24, 0xD8	; 216
 320:	80 93 71 00 	sts	0x0071, r24	; 0x800071 <__DATA_REGION_ORIGIN__+0x11>
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:53
  WDTCSR &= ~(1<<WDE);
 324:	81 b5       	in	r24, 0x21	; 33
 326:	87 7f       	andi	r24, 0xF7	; 247
 328:	81 bd       	out	0x21, r24	; 33

The strange thing is that when I’m setting up and enabling the watchdog as the last lines in setup, the OR’ing of the bits in register WDTCSR is working as expected

void setup_watchdog() {
  cli();
 3ba:	f8 94       	cli
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:83
  MCUSR &= ~(1<<WDRF);
 3bc:	84 b7       	in	r24, 0x34	; 52
 3be:	87 7f       	andi	r24, 0xF7	; 247
 3c0:	84 bf       	out	0x34, r24	; 52
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:84
  WDTCSR &= ~(1<<WDIE);
 3c2:	81 b5       	in	r24, 0x21	; 33
 3c4:	8f 7b       	andi	r24, 0xBF	; 191
 3c6:	81 bd       	out	0x21, r24	; 33
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:85
  CCP = 0xD8;
 3c8:	88 ed       	ldi	r24, 0xD8	; 216
 3ca:	80 93 71 00 	sts	0x0071, r24	; 0x800071 <__DATA_REGION_ORIGIN__+0x11>
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:86
  WDTCSR |= ((1<<WDE) | (1<<WDP3) | (0<<WDP2) | (0<<WDP1) | (0<<WDP0));
 3ce:	81 b5       	in	r24, 0x21	; 33
 3d0:	88 62       	ori	r24, 0x28	; 40
 3d2:	81 bd       	out	0x21, r24	; 33
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:88
  sei();
 3d4:	78 94       	sei
C:\Users\Steen\Documents\Elektronik\Arduino\Projects\WDT_test/WDT_test.ino:89
  wdt_reset();
 3d6:	a8 95       	wdr

Am I doing something wrong, or do the AtTinycore have problems with the AND function when it is using a value of several registers OR’ed together ?

Br
Steen

sleemanj, 2022-04-12:
> WDTCSR &= ((0<<WDIE) | (0<<WDE));
> 31e: 81 b5 in r24, 0x21 ; 33
> 320: 11 bc out 0x21, r1 ; 33

Notice a couple things, first, two different registers, `r24` and `r1`, second is that the right hand side of the expression is equivalent to 0.

Your subsequent attempts at "negative logic" produce the same result, the right hand side evaluates to 0

`0 & anything` is 0 so the only thing to compiler has to do there as far as I can see is set WDTSCR to 0.

`r1` is I believe a zero register, it's always 0.  So that is just the compiler doing exactly what you asked, set WDTCSR to 0.

It may "work" in your "split" way simply because the optimiser doesn't see it the same.   But end of the day, you're still setting every bit to 0.

I think maybe you mean to use `|=` instead of `&=`.

See also the timed procedure in the datasheet for setting WDE.

SpenceKonde, 2022-04-15:
Quoth the datasheet:
_In this mode, the Watchdog Timer is initially disabled, but can be enabled by writing the WDE bit to one without
any restriction. A timed sequence is needed when disabling an enabled Watchdog Timer. To disable an enabled
Watchdog Timer, the following procedure must be followed:
1. Write the signature for change enable of protected I/O registers to register CCP
2. Within four instruction cycles, in the same operation, write WDE and WDP bits_
I'm amazed it's working in any of those cases because you generally have more than 4 instruction cycles there. 

Just use straight assignment to set the WDT register, since you'll typica;;y be setting all bits at once. Using |= causes a read-modify-write which takes more than 4 instruction clocks

closed 2022-04-15 by SpenceKonde