SpenceKonde/ATTinyCore #691: Arduino Board Manager errors when installing version 1.5.0 and up

nbaddorf, 2022-05-15:
Hi, I just tried to install the AttinyCore board in arduino, I am using "http://drazzy.com/package_drazzy.com_index.json" as the link. When I select version 1.5.0 or 1.5.2, it gives me this error: 
`Error downloading
https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86 64-apple-darwin.tar.bz2`

Just thought I should point this out incase anyone else has issues. Version 1.4.1 is working great for me.

The arduino terminal box or what ever it is called at the bottom of a sketch gives this info:
`Error downloading https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-apple-darwin.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Error downloading https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-apple-darwin.tar.bz2
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-apple-darwin.tar.bz2
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:83)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:113)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:198)
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1967)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:331)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:325)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1688)
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:226)
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:1082)
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:1010)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1079)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1388)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1416)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1400)
	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.connect(HttpsURLConnectionImpl.java:167)
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:155)
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:106)
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:234)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
	... 5 more
Caused by: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed
	at sun.security.validator.PKIXValidator.doValidate(PKIXValidator.java:380)
	at sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:273)
	at sun.security.validator.Validator.validate(Validator.java:262)
	at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:330)
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:237)
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:132)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1670)
	... 21 more
Caused by: java.security.cert.CertPathValidatorException: validity check failed
	at sun.security.provider.certpath.PKIXMasterCertPathValidator.validate(PKIXMasterCertPathValidator.java:135)
	at sun.security.provider.certpath.PKIXCertPathValidator.validate(PKIXCertPathValidator.java:233)
	at sun.security.provider.certpath.PKIXCertPathValidator.validate(PKIXCertPathValidator.java:141)
	at sun.security.provider.certpath.PKIXCertPathValidator.engineValidate(PKIXCertPathValidator.java:80)
	at java.security.cert.CertPathValidator.validate(CertPathValidator.java:292)
	at sun.security.validator.PKIXValidator.doValidate(PKIXValidator.java:375)
	... 27 more
Caused by: java.security.cert.CertificateExpiredException: NotAfter: Sun May 15 07:45:07 EDT 2022
	at sun.security.x509.CertificateValidity.valid(CertificateValidity.java:274)
	at sun.security.x509.X509CertImpl.checkValidity(X509CertImpl.java:677)
	at sun.security.provider.certpath.BasicChecker.verifyValidity(BasicChecker.java:190)
	at sun.security.provider.certpath.BasicChecker.check(BasicChecker.java:144)
	at sun.security.provider.certpath.PKIXMasterCertPathValidator.validate(PKIXMasterCertPathValidator.java:125)
	... 32 more
 Error downloading https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-apple-darwin.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Error downloading https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-apple-darwin.tar.bz2
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-apple-darwin.tar.bz2
[... 87 lines cut ...]
Caused by: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed
	at sun.security.validator.PKIXValidator.doValidate(PKIXValidator.java:380)
	at sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:273)
	at sun.security.validator.Validator.validate(Validator.java:262)
	at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:330)
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:237)
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:132)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1670)
	... 21 more
Caused by: java.security.cert.CertPathValidatorException: validity check failed
	at sun.security.provider.certpath.PKIXMasterCertPathValidator.validate(PKIXMasterCertPathValidator.java:135)
	at sun.security.provider.certpath.PKIXCertPathValidator.validate(PKIXCertPathValidator.java:233)
	at sun.security.provider.certpath.PKIXCertPathValidator.validate(PKIXCertPathValidator.java:141)
	at sun.security.provider.certpath.PKIXCertPathValidator.engineValidate(PKIXCertPathValidator.java:80)
	at java.security.cert.CertPathValidator.validate(CertPathValidator.java:292)
	at sun.security.validator.PKIXValidator.doValidate(PKIXValidator.java:375)
	... 27 more
Caused by: java.security.cert.CertificateExpiredException: NotAfter: Sun May 15 07:45:07 EDT 2022
	at sun.security.x509.CertificateValidity.valid(CertificateValidity.java:274)
	at sun.security.x509.X509CertImpl.checkValidity(X509CertImpl.java:677)
	at sun.security.provider.certpath.BasicChecker.verifyValidity(BasicChecker.java:190)
	at sun.security.provider.certpath.BasicChecker.check(BasicChecker.java:144)
	at sun.security.provider.certpath.PKIXMasterCertPathValidator.validate(PKIXMasterCertPathValidator.java:125)
	... 32 more
`

SpenceKonde, 2022-05-15:
Are you fucking serious?

Auto update was turned on

email alerts when cert expires were turned on.

No email alerts

No automatic update

And LetsEncrypt, theonly place where you don't have to spend and arm, a leg
and firstborn son (I don't have any kids, so the other places would
probable have to chop off something else so I never had one) to get a
fucking SSL cerl keeps us on a stjupid short leash, like 30 fucking days.
And now apparently doesn't even fucking tell me "HEY WE'RE ABOUT TO TAKE
DOWN YOUR WEBSITE IF YOU DON'T UPDATE THE CERT!" 24 hours before it does so
like it used to

I fucking hate SSL and I fucking hate LetsEncrypt. I am furious.

Nobody, anywhere, couldi install the core.

Because LetsEncerypt now **SILENTLY** expires certs.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

SpenceKonde, 2022-05-15:
This was impacting ALL versions that supported micronucleus because the assets are hosted on Azduino,com.

SpenceKonde, 2022-05-15:
It is fixed now. And I am furious at LetsEncrypt.

SpenceKonde, 2022-05-15:
You may have to restart Arduino or reboot the computer or clear the IE cache to clear. 

And all this just because security nutjobs give me hell if I host anything on a non-SSL server.

closed 2022-05-15 by SpenceKonde