SpenceKonde/ATTinyCore #693: V2.0.0 wrong fuse setting when "burn bootloader" for Attiny2313

hmeijdam, 2022-05-30:
After being confused for an hour why my sketch was running at the wrong speed I discovered that on my installation of V2.0.0 I am getting wrong fuse settings when using burn bootloader.

When selecting "8MHz internal"  I am getting LFuse 0xE2 , HFuse 0xD7 and EFuse 0xFE. This looks wrong to me (4MHz and a lot of other wrongs) 

When selecting the same "8MHz internal" on my boardsmanager installation of ATTinycore I get  LFuse 0xE4 , HFuse 0x9F and EFuse 0xFF. That seems OK to me.

I stopped testing other options, because if the HFuse 0xD7 would have been a 0xD6 my reset pin would have been disabled and then I need to hook up a HV programmer first.

Maybe worth to notice but if I select an ATTiny85 and do "burn bootloader" for 8 or 12 MHz (internal tuned) I am getting the exact same fuse settings in the ATTiny85: LFuse 0xE2 , HFuse 0xD7 and EFuse 0xFE

So now I wonder; do I have some corrupted file just on my PC, or can someone test if he get's the same result?

hmeijdam, 2022-05-30:
I looked in both boards.txt files and can find where the difference is coming from.

AT2313 Boards.txt in github version V2.0.0

```

################################
# Clocking menu options        #
################################
attinyx41.menu.clock.internal_8m_4v5=8 MHz (internal > 4.5V)
attinyx41.menu.clock.internal_8m_4v5.bootloader.low_fuses=0xE2
attinyx41.menu.clock.internal_8m_4v5.build.f_cpu=8000000UL
attinyx41.menu.clock.internal_8m_4v5.build.speed=8m
attinyx41.menu.clock.internal_8m_4v5.build.clocksource=0
attinyx41.menu.clock.internal_8m_4v5.build.tuneorcal=-DCAL_NUDGE_DOWN
attinyx41.menu.clock.internal_8m=8 MHz (internal)
attinyx41.menu.clock.internal_8m.bootloader.low_fuses=0xE2
attinyx41.menu.clock.internal_8m.build.f_cpu=8000000UL
attinyx41.menu.clock.internal_8m.build.speed=8m
attinyx41.menu.clock.internal_8m.build.clocksource=0
```

boards.txt from boardsmanager version of ATTinycore
```

attinyx313.name=ATtiny2313(a)/4313 (No bootloader)
attinyx313.upload.maximum_size=4096
attinyx313.upload.maximum_data_size=256
attinyx313.upload.tool=avrdude
attinyx313.bootloader.tool=avrdude
attinyx313.bootloader.unlock_bits=0xFF
attinyx313.bootloader.lock_bits=0xFF
attinyx313.bootloader.file=empty/empty_all.hex
attinyx313.build.core=tiny
attinyx313.build.board=AVR_ATTINYX313
attinyx313.build.export_merged_output=false
attinyx313.bootloader.extended_fuses=0xFF
attinyx313.menu.LTO.enable=Enabled
attinyx313.menu.LTO.enable.ltocflags=-flto -fno-fat-lto-objects
attinyx313.menu.LTO.enable.ltoelfflags=-g -flto -fuse-linker-plugin
attinyx313.menu.LTO.enable.ltocppflags=-flto
attinyx313.menu.LTO.enable.ltoarcmd=gcc-
attinyx313.menu.LTO.disable=Disabled
attinyx313.menu.LTO.disable.ltocflags=
attinyx313.menu.LTO.disable.ltoelfflags=
attinyx313.menu.LTO.disable.ltocppflags=
attinyx313.menu.LTO.disable.ltoarcmd=
attinyx313.menu.chip.4313=ATtiny4313
attinyx313.menu.chip.4313.build.mcu=attiny4313
attinyx313.menu.chip.4313.upload.maximum_size=4096
attinyx313.menu.chip.4313.upload.maximum_data_size=256
attinyx313.menu.chip.2313=ATtiny2313/ATtiny2313A
attinyx313.menu.chip.2313.build.mcu=attiny2313a
attinyx313.menu.chip.2313.upload.maximum_size=2048
attinyx313.menu.chip.2313.upload.maximum_data_size=128
attinyx313.menu.clock.8internal=8 MHz (internal)
attinyx313.menu.clock.8internal.bootloader.low_fuses=0xE4
attinyx313.menu.clock.8internal.build.f_cpu=8000000L
attinyx313.menu.clock.8internal.build.clocksource=0
```

SpenceKonde, 2022-05-31:
Thanks!

Actually the 828 does have E2 as low fuse, depending on how you count the reserved bits. They don't seem to make a difference. 

Found the issues with the boards.txt build script - I had to give it a whole second set of internal speeds for the x313. 
My apologies for the fate of the boards that got reset disabled. Hadn't realized the HFuse had weird layout on the x313s

SpenceKonde, 2023-06-25:
HAs long since been addressed

closed 2023-06-25 by SpenceKonde