SpenceKonde/ATTinyCore #695: Bootloader (optiboot) and MAX485 RXTX pin

expertup, 2022-06-12:
I have Attiny85 and MAX485. I want to programming attiny by 485. I know that optiboot can change max485 in to "programming mode" by pin from attiny.

Is this option with pin isenabled default?

Or how compile own version? I tryied compile, but I have on any step error. I does this on windows and ubuntu. Is any tutorial step by step?

SpenceKonde, 2022-06-15:
Did you read https://github.com/SpenceKonde/ATTinyCore/discussions/694?

Also, try using the 2.0.0-dev branch - all new development is happening there

expertup, 2022-06-17:
RS485 TXRX dicerction communication need resistor pull up/down? In RS485=B2 pin level is very unstable and jumping. On about 2 weeks I wil have osciloscope.

```
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Dell\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/avrdude.conf"

         Using Port                    : COM8
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
         Setting bit clk period        : 5.0
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x0d
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x0d
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x0d
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x0d
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x0d
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x0d
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x0d
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x0d
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x0d
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x0d

avrdude done.  Thank you.

Problem uploading to board.  See https://support.arduino.cc/hc/en-us/sections/360003198300 for suggestions.

```

SpenceKonde, 2022-07-04:
I don't see anything obviously wrong in the code -  but I never used the RS485 stuff myself.... that was added for someone else and declared by them to be working, then years later bill westfield submitted a PR that said rs485 was actually broken and supposedly fixing it. Nobody has ever tested it since, I don't think, and we do not distribute compiled binaries. But just looking at the size of the compiled binary, it doesn't look like the feature is taking up any space, implying that the option is getting lost. 

I also can't recommend using such brand new versions of avr-gcc, as AVR is not watched as closely as say, x64 or ARM. Microchip uses 7.3 IIRC, same thing as arduino,, and nobody has their eye on the later gcc branches to make sure AVR has not regressed. It certainly doesn't seem to have progressed signficantly, and it, in effect, targets x64 and ARM and stuff like that increasingly, and offer little benefit to AVR, and nobody's even making sure they;re not accumulating mold (like that spurrious warning about array bounds which assumes 0 isn't a valid memory address (true on many platforms. Not true on AVR, and there's apparently a simple way to fix it, but nobody has jumped through the hoops needed to get that configuration setting for AVR changed in several versions of avr-gcc. My understanding is that most people use 7.3.

SpenceKonde, 2023-06-20:
Okay, I'm going to correct what I said previously - for Optiboot bootloaders, we use the 5.4.0 toolchain as we have gotten binaries that did not work correctly out of later ones. Adding support for new toolchain versions for legacy parts is not a priority.