SpenceKonde/ATTinyCore #697: Interrupt

gioreva, 2022-06-17:
Interrupt not work.

Tiny85
Button on PB0 (Work on polling)

attachInterrupt(digitalPinToInterrupt(0), VoidInt, RISING);
attachInterrupt(0, VoidInt, RISING);
attachInterrupt(digitalPinToInterrupt(0), VoidInt, LOW;
attachInterrupt(0, VoidInt, LOW;

void VoidInt(void){
  if( ckOn ) resOn;
  else setOn;
}

gioreva, 2022-06-17:
Also SLEEP seems that it doesn't work.

Always absorbs 4.5mA

set_sleep_mode(SLEEP_MODE_PWR_DOWN);
sleep_enable();
delay(100);
sleep_cpu();
sleep_disable();

gioreva, 2022-06-17:
More info, This my include:

#include <avr/sleep.h>
#include <avr/interrupt.h>
#include <avr/wdt.h>
#include <avr/io.h>
#include <avr/power.h>
#include <util/atomic.h>

gioreva, 2022-06-17:
If i use

MCUCR = (1 << SE) | (1 << SM1);   powerdown

instead of

set_sleep_mode(SLEEP_MODE_PWR_DOWN);

Sleep work well.

SpenceKonde, 2022-06-23:
So - my diagnosis is that the attachInterrupt issue is user error, that pin does not support attaching interrupts and the Arudino API does not support pcints. 

That the Arduino API doesn't wrap PCINTS is something we should all be thankful for.  On modern AVRs they got rid of the INTn interrupts, giving most of their features to "pin interrupts" which acted much like PCINTs in banks of 8. Great! But the clowns who write the official Arduino core had to make attachInterrupt work on all pins.... so it has logic to pick which of the up-to-8 interrupts to call. 

The end result? If any user code or library referenced attachInterrupt(), it would pull in an ISR for EVERY PIN INTERRUPT, and you could no longer make one that executed in 8 or even 19 clock cycles, but rather one that executed in 50-100 clocks, even for an empty interrupt, AND because it glommed onto every pin vector, you can't even manually define an interrupt that will execute quickly. It is an unholy abomination.

attachInterrupt has the same latency problem by the way, though not as severely. 

```
MCUCR |= MCUCR & ((1<<ISC1) | (1 <<ISC0)) | (0, 1, 2 or 3);
// 0 = LOW LEVEL, 1 = any edge, 2 = falling edge 3 = rising edge. 

ISR(some_vect) {
  /* code, but no function calls that won't be inlined */
}
````
is up to 29 clocks slower to enter and 28 clocks slower on exit than
```
attachInterrupt(myisr, (LOW, RISING, FALLING or CHANGE));
// This is actually probably understating the inefficiency - it doesn';t
void myisr() {
  /* the same exact code as the interrupt above */
}
```
because to comply with C the ABI, it has to push every call used register onto the stack (2 clocks per register, and there are 14 of them - but r0 and r1 (temp_reg and zero_reg) get pushed for any non-EMPTY_INTERRUPT. non-NAKED_INTERRUPT), load two bytes into the Z register and icall that function pointer (5 clocks), then at the end, restore all those values by popping them off the stack.  (says right in the datasheet not to call functions from within ISRs because of the performance impact...)

On the modern AVRs. I eventually got so f'ing fed up with it, because libraries would use attachInterrupt and then you couldn't manually define pin interrupts, that as of mTC 2.5.0 and DxC 1.4.0, I replaced the 1 file with up to 7 ISRs (for each port present on the part) with 1 file per port, so they wouldn't get pulled in unless something referenced something else in that file, and then put in trivial functions attachAEnable() attachBenable() which would get optimized away because they were do-nothing functions, but were sufficient to pull in the ISR, and replaced each ISR with a NAKED_ISR that pushed a single register onto the stack (to make room), then loaded 2*port number via LDI into that register, and then called a ((naked))((used)) function defined back in WInterrupts (since everything is naked, no pushing or popping was done automatically), and I then manually push r0, load and push the status register, clear r1. and push the necessary registers to call a function, plus the one needed to make the pin detection logic work, implemented that in assembly to loop over the port's interrupt flags, and icall the appropriate pointer (unless it was a null pointer), and then restore the call used registers and SREG and return from the interrupt. 

While 2.5.0/1.4.0 worked in my easy test, It took until 2.5.11 and 1.4.10 before it worked correctly for all cases. Those cores have a tools menu to select either the new implementation, but with all the attach enable functions called at startup, (the default), only enable the interrupts that the user manually enables, or fall back to the old version (because I realized the chance that it would actually work on intial release was just about 0%). I did something very similar to that for the USART interrupts too not for speed (because there wasn't an advantage there) but because it avoided duplicating the ISR code for each USART (and DA/DB parts have up to 6, while tinyAVR parts can have 2 USARTs on parts with as little as 4k of flash.

gioreva, 2022-06-23:
I have problems understanding everything you wrote, both for my bad English and because I have very little time.

My project now works well in this way.

`MCUCR = (1 << SE) | (1 << SM1);     //  set_sleep_mode(SLEEP_MODE_PWR_DOWN);`

This only for wake up:
```
  PCMSK = 1;  // Pin change interrupt    (1 << PCINT0)
  GIMSK = (1 << PCIE);    // Pin change enable
```

This not work, i have lost 20uA
Marking of my chip is T85 20UBRK 11918M
If B is relise code, need relise from C 
```
  noInterrupts();
  byte reg1 = MCUCR | (1<<BODS) | (1<<BODSE);
  byte reg2 = MCUCR & ~(1<<BODSE);
  MCUCR = reg1;
  MCUCR = reg2;
  interrupts();
```

```
void Off ( void ){
  analogReference(DEFAULT);
  ClearBit(ADCSRA,ADEN);    // ADC off,  this save 200uA
  resOn;
  sleep_cpu();  
  SetBit(ADCSRA,ADEN);      // ADC  on
  analogReference(INTERNAL);
}
ISR(PCINT0_vect){  // Pin Change interrupt
}
```

Total consumption is 20uA with on BOD, and 0uA with off BOD

closed 2022-06-30 by SpenceKonde