SpenceKonde/ATTinyCore #702: tinyNeoPixel_Static issue with atmega168

atghrr, 2022-07-01:
Hello! The Library [ATTinyCore](https://github.com/SpenceKonde/ATTinyCore)/[avr](https://github.com/SpenceKonde/ATTinyCore/tree/master/avr)/[libraries](https://github.com/SpenceKonde/ATTinyCore/tree/master/avr/libraries)/[tinyNeoPixel_Static](https://github.com/SpenceKonde/ATTinyCore/tree/master/avr/libraries/tinyNeoPixel_Static)/tinyNeoPixel_Static.cpp works with atmega 168 at 16MHz, but there is compilation error when MC clocks at 8MHz. (using [MCUdude](https://github.com/MCUdude) /[MiniCore](https://github.com/MCUdude/MiniCore))

I tried to change "NEOPIXELPORT" with just "port", but received another error with asm:

```
...tinyNeoPixel_Static.cpp: In member function 'void tinyNeoPixel::show()':
tinyNeoPixel_Static.cpp:229:23: error: impossible constraint in 'asm'
     [lo]     "r" (lo));
```

Please help, because the library is tiny and useful - i liked it.

SpenceKonde, 2022-07-01:
I will intestate whether there us a solution.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2022-07-04:
I cannot reproduce this with the latest version of the library (from v2.0.0-dev branch).  When I started reading the code as someone who knew avr assembly, I realized there were piles of issues, first and foremost being that it had to be told what port to use to avoid duplicating code (which we don't want to waste flash on). The comments said it wasn't possible to meet the timing specs at 8 MHz using the 'st' instruction (which takes 2 cycles on classic AVRs to store to a memory address pointed to by the X, Y, or Z register; (more recent parts execute it in 1), so they wrote the port register with the 1-cycle `out` instruction. But that must use a compile time known address, so the port had to be known at compile time, and since C++ seems to kneecap the optimizer, even if only one port is used with the library, the compiler can't perform constant folding, hence the tinyneopixel menu. 

It turns out that with more aggressive programming practices, it was entirely possible to make the asm run fast enough with st, and hence without a tools submenu. 
The library also only worked at all because the optimizer gets confused by C++, and hence would always pick registers that were legal for the actual instruction, even though the constraints were not as strict as they had to be, and because memory accesses to get the pixel pointer were not optimized away for the same reason, which was a damned good thing, because we told the compiler that the assembly wouldn't change the value of the pointer, when we increment it 3 times per pixel. And some speeds use a bit counter, specified as "any register", but it gets initialized to 8 at the start of every byte with ldi, which only works on "upper registers". The register allocation algorithm just happened to always give it an upper register to that. So try with the v2.0.0-dev branch version of the library, it should work. If it doesn't, post the full text output with "Show verbose output during compile" checked in preferences.

closed 2022-07-04 by SpenceKonde