SpenceKonde/ATTinyCore #705: 'NEOPIXELPORT' was not declared in this scope

corr0si0n, 2022-07-14:
When trying to compile the following sketch

```
#include <tinyNeoPixel_Static.h>
#define PIN 1
#define NUMPIXELS 4
byte pixels[NUMPIXELS * 3];
tinyNeoPixel strip = tinyNeoPixel(NUMPIXELS, PIN, NEO_GRB, pixels);

void setup() {
  pinMode(PIN, OUTPUT);
}

void loop() { }
```

I get the error:
```
/Users/r00/Documents/Arduino/libraries/tinyNeoPixel_Static/tinyNeoPixel_Static.cpp: In member function 'void tinyNeoPixel::show()':
/Users/r00/Documents/Arduino/libraries/tinyNeoPixel_Static/tinyNeoPixel_Static.cpp:137:8: error: 'NEOPIXELPORT' was not declared in this scope
   hi = NEOPIXELPORT |  pinMask;
        ^~~~~~~~~~~~
/Users/r00/Documents/Arduino/libraries/tinyNeoPixel_Static/tinyNeoPixel_Static.cpp:137:8: note: suggested alternative: 'NOT_A_PORT'
   hi = NEOPIXELPORT |  pinMask;
        ^~~~~~~~~~~~
        NOT_A_PORT
exit status 1
Error compiling for board ATtiny25/45/85.
```

I'm using the ATtiny85 board core [here](https://github.com/damellis/attiny). Is this a bug in the library (doubt it) or do I need to change my configuration?

hmeijdam, 2022-07-14:
With AttinyCore it compiles just fine. So suggest you change to AttinyCore.

I am puzzled why you log an issue in for AttinyCore, while you are using a different core.

sleemanj, 2022-07-14:
That damellis "core" should die a horrible death.  See so many people trying to use it with ATtinys.  It has no core in it, it's just a boards defintion to use the Arduino core and hope for the best.

SpenceKonde, 2022-07-14:
A surprising number of prople do it, particularly with the recommended
_static versionThe reliance on the neopixelport define Iis gone in
2.0.0-dev branch, use version of that library from there (other parts pf
core are busted so can't release) but the new tinyneopixel (static and
normal) should be fully compatible between attinycore 2.0.0-dev and any
classic avr (and between megaTinyCore or dxcore (they share the same code
exactly for that lib) and any modern avr)

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

closed 2022-07-14 by corr0si0n