SpenceKonde/ATTinyCore #707: Interrupt pins on the Attiny88/MH-ET-Tiny88?

IdontKnowWhatToNameThisAccount, 2022-07-21:
Which of the pins on the MH-ET-Tiny88 board are interrupt pins and what is the code to get these to work? I cant find any info on that topic although the datasheet for the Attiny88 mentions the word "Interrupt" 681 times and even mentions external interrupts. So how do these work in this library?

SpenceKonde, 2022-07-21:
attachInterrupt works on the INT0 and INT1 pins (2 and 3) and should work like it does on a normal classic AVR. 
All pins can be used as pin-change interrupts - this must be implemented manually (the limitations of the C/arduino framework impose a significant performance penalty on any sort of "attached" interrupt, because right off the bat, you lose like 30 clock cycles entering and exiting the ISR saving and restoring registers, because the optimizer doesn't know what function will be called and hence can't inline it, and has to free up all 16 call used registers, instead of just r1 and r0. and restore them afterwards; Often a well-written (ie, short) manually defined interrupt can run and return to the code that was interrupted by the time that one created with an attachInterrupt like function would have been done with the prologue. 

In general, barring the pin numbering differences after the first 14 pins or so, interrupts should work identically to a 328p - they tried to keep some semblance of code compatibility on the tiny x8-family with the mega x8 family (note also how there's both a real TWI and a real SPI port - on a low cost tiny, instead of that dreadful USI (Unwieldy Serial Interface`*`) that most tinyAVRs before the 2016 revolution got stuck with)- very un-Atmel like of them, most tinyAVRs from that era likl like they chose pin mappings and special functions by throwing darts at a pinout chart while blindfolded. Microchip seems to have whipped them into shape on that front and the modern AVRs have much more consistent pin mappings and pinouts. 

`*` I think they called it "universal serial interface" but that does not describe it's functionality very well.

IdontKnowWhatToNameThisAccount, 2022-07-22:
Sorry, I only understood maybe an eighth of what you wrote. Im looking for a more specific example. If I understood you right, The Pins labeled as 2 and 3 on the MH-ET board actually have interrupts build in, right? But what would I need to do to get them to work? I have a line of code for an arduino nano I'd like to adopt:
`attachInterrupt(digitalPinToInterrupt(pin), impulse, FALLING);`
Ive tried just replacing "pin" with "3" but that doesnt seem to work. As I understand your text digitalPinToInterrupt is bad anyways, but whats the alternative? Ive tried pin 2 too but that seems to interfere with usb functionality

SpenceKonde, 2022-08-04:
I think what you want is 
```
attachInterrupt(digitalPinToInterrupt(PIN_PD3), myEventListener, CHANGE);
```

SpenceKonde, 2022-08-04:
(PIN_PD2, AKA INT0 is generally one of the USB pins - the one that needs to respond fastest. Since, when multiple interupts are enabled and and their flag is set, if you want a prayer of USB workingf in an interrupt driven configuration,m int0 must be used for USB if you want to potentially use USB from the sketch (which still is super flaky)

If you just need an event on a *change* not rising/falling/level, then you can just use a PCINT, which can be used on any pin (though there's only one vector per portso things get a bit more complicated if you want more than one interrupt pin on a given port, but usually you don't do that unless it makes things more convenient rather than less

closed 2022-09-09 by SpenceKonde