SpenceKonde/ATTinyCore #711: ATtiny88 Micronucleus Upload Error

SpenceKonde, 2022-08-03:
### Discussed in https://github.com/SpenceKonde/ATTinyCore/discussions/710

<div type='discussions-op-text'>

<sup>Originally posted by **founfabug** August  3, 2022</sup>
Arduino: 1.8.13
ATTinyCore-2.0.0-dev: From today
System: Windows 10
Error on uploading:
`java.io.IOException: Cannot run program "C:\Users\...\AppData\Local\Arduino15\packages\ATTinyCore\tools\micronucleus\2.5-azd1b/launcher": CreateProcess error=2, The system cannot find the file specified`

`micronucleus.exe` is at that location.

Note that `platform.txt` contains a definition: 
`tools.micronucleus.cmd.path={runtime.tools.micronucleus.path}/launcher` 
which differs from master branch: 
`tools.micronucleus.cmd.path={runtime.tools.micronucleus.path}/micronucleus `

Reverting the definition in `platform.txt` resolves the error, but introduces `Unrecognized option: -cdigispark`

Additional debugging indicates other things are not quite right, including my understanding of what is going on :-)

Uploads as expected with 1.5.2 core.

_Thank you_</div>

SpenceKonde, 2022-08-03:
This is one of the many reasons 2.0.0-dev is 2.0.0-dev instead 0f 2.0.0

SpenceKonde, 2022-08-03:
I wonder if this is reklated to the issues I was having with the 861 ultramini boards I'm planning to sell as soon as I can make the bloody things work

ArminJo, 2022-08-26:
If I use your core I get the error:
`Error during install: Error downloading tool ATTinyCore:micronucleus@2.5-azd1b: Get "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-linux-gnu.tar.bz2": dial tcp 3.218.2.136:443: connect: connection timed out`
see here:
https://github.com/ArminJo/Github-Actions/runs/8044312905?check_suite_focus=true#step:4:312)
and here
https://github.com/ArminJo/Arduino-OpenWindowAlarm/runs/8044572270?check_suite_focus=true#step:3:63

SpenceKonde, 2022-09-09:
I can't reproduce that issue, it sounds like a connectivity issue of some sort....

ArminJo, 2022-09-09:
I thought you fixed it, since it worked a few hours later 😏

SpenceKonde, 2022-09-09:
All I did was log onto the server long enough to do ps to see that httpd was runing, and confirm that I could download it. 

Okay, 2.0.0 is now calling micronucleus directly for uploads.

closed 2022-09-09 by SpenceKonde